RfidResearchGroup/proxmark3 #2119: lf em 4x05 chk

jareckib, 2023-10-06:
Password search doesn't work. Each time you run the command, you will get different passwords found. 
None of the found passwords work.

iceman1001, 2023-10-06:
You would need to fill in some more details in order to look into what is not working....

jareckib, 2023-10-07:
There's nothing to explain. I have several Em4305 with a password imposed by a Chinese duplicator. After running the command ---lf em 4x05 chk... it finds the passwords from the file, or it does not find them... Unfortunately, the ones found do not work either....19924027....or 11111111 or ABCD1234 or 44444444 etc.

iceman1001, 2023-10-08:
Not really constructive feedback.  

Your opinion on the commands is most likely best expressed in the discord server. 

Closing.

closed 2023-10-08 by iceman1001