RfidResearchGroup/proxmark3 #2122: Mifare Plus X 4K (SL1) hardnested fails on sector 32 and above

Myppomeu, 2023-10-10:
Attempt of hardnested attack for sector 32 and above on Mifare Plus X 4K in SL1 fails:

```
[=] Target block no 143, target key type: B, known target key: 000000000000 (not set)
[=] File action: none, Slow: No, Tests: 0
[=] Hardnested attack starting...
[=] ---------+---------+---------------------------------------------------------+-----------------+-------
[=]          |         |                                                         | Expected to brute force
[=]  Time    | #nonces | Activity                                                | #states         | time
[=] ---------+---------+---------------------------------------------------------+-----------------+-------
[=]        0 |       0 | Start using 4 threads and AVX2 SIMD core                |                 |
[=]        0 |       0 | Brute force benchmark: 1190 million (2^30,1) keys/s     | 140737488355328 |   33h
[=]        1 |       0 | Loaded 0 RAW / 351 LZ4 / 0 BZ2 in 520 ms                | 140737488355328 |   33h
[=]        1 |       0 | Using 239 precalculated bitflip state tables            | 140737488355328 |   33h
[#] AcquireEncryptedNonces: Auth2 error len=1
[#] AcquireEncryptedNonces: Auth2 error len=1
[#] AcquireEncryptedNonces: Auth2 error len=1
[#] AcquireEncryptedNonces: Auth2 error len=1
[#] AcquireEncryptedNonces: Auth2 error len=1
...
[#] AcquireEncryptedNonces: Auth2 error len=1
[#] AcquireEncryptedNonces: Auth2 error len=1

[!!]  Error: No response from Proxmark3

[!]   Communicating with Proxmark3 device failed

[=] Running in OFFLINE mode. Use "hw connect" to reconnect
```

Unfortunately I don't have any Mifare Classic 4K card to check if bug can be reproduced with it.

**To Reproduce**
For example:
```
hf mf hardnested -b -k FFFFFFFFFFFF --blk 29 --tblk 143 --tb
hf mf autopwn --4k -b -k FFFFFFFFFFFF -s 7 --slow
```

**Desktop:**
 - Debian 11, Proxmark3 Easy from PiSwords
 
```
[usb|script] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ Client ]
  Iceman/master/v4.17140-suspect 2023-09-09 11:00:00 5901f2664
  compiled with............. GCC 10.2.1 20210110
  platform.................. Linux / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... present
  Python script support..... present
  Lua SWIG support.......... present
  Python SWIG support....... present

 [ Proxmark3 ]
  firmware.................. PM3 GENERIC

 [ ARM ]
  bootrom: Iceman/master/v4.17140-suspect 2023-09-09 11:00:00 5901f2664
       os: Iceman/master/v4.17140-suspect 2023-09-09 11:00:00 5901f2664
  compiled with GCC 8.3.1 20190703 (release) [gcc-8-branch revision 273027]

 [ FPGA ]
  fpga_pm3_lf.ncd image 2s30vq100 2023-08-29 16:44:07
  fpga_pm3_hf.ncd image 2s30vq100 2023-08-29 16:44:19
  fpga_pm3_felica.ncd image 2s30vq100 2023-08-29 16:44:43
  fpga_pm3_hf_15.ncd image 2s30vq100 2023-08-29 16:44:31

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 60% used )
```

```
[usb|script] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 42552
[#]   Available memory........ 42552
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... fpga_pm3_hf.ncd image 2s30vq100 2023-08-29 16:44:19
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0
[#]
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
[#]     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
[#]               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A |
[#]    1 of 4 coding reference |  31 |  20 |  18 |  34 |  15 |  50 |  66 |
[#]
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 283136
[#]   Transfer Speed PM3 -> Client... 566272 bytes/s
[#] Various
[#]   Max stack usage......... 4104 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... -1
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 31154 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[#]
```

```
[usb|script] pm3 --> data tune
[=] ---------- Reminder ------------------------
[=] `hw tune` doesn't actively tune your antennas,
[=] it's only informative.
[=] Measuring antenna characteristics, please wait...

[=] ---------- LF Antenna ----------
[+] LF antenna: 25,23 V - 125,00 kHz
[+] LF antenna: 17,78 V - 134,83 kHz
[+] LF optimal: 25,68 V - 122,45 kHz
[+] Approx. Q factor (*): 6,6 by frequency bandwidth measurement
[+] Approx. Q factor (*): 7,5 by peak voltage measurement
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 15,08 V - 13.56 MHz
[+] Approx. Q factor (*): 4,4 by peak voltage measurement
[+] HF antenna is OK

(*) Q factor must be measured without tag on the antenna

[+] Displaying LF tuning graph. Divisor 88 (blue) is 134,83 kHz, 95 (red) is 125,00 kHz.
```

Myppomeu, 2023-10-10:
As I hear, this software uses mfoc-hardnested code. [Related issue](https://github.com/nfc-tools/mfoc-hardnested/issues/10) from it's repo with PR.

iceman1001, 2023-10-10:
Mifare Plus  != Mifare Classic

Myppomeu, 2023-10-10:
Hardnested should not work? First 32 sectors subject to attack.

iceman1001, 2023-10-10:
Different card tech which allows for different access rights to the card memory.

This isn't a issue of the source code, 
The right place to ask questions is in the discord server.

closed 2023-10-10 by iceman1001