RfidResearchGroup/proxmark3 #2141: Connection failure issue occurs when using Proxmark from CLI

dhje0ng, 2023-10-19:
**Describe the bug**
A connection error occurs when using commands in CLI mode with Proxmark3 ICEMAN firmware.
In detail, when the first command is sent in CLI mode, it is connected to the hardware normally.

If you subsequently try to connect a second time, an error will be displayed indicating that communication with Proxmark has failed. Is there any solution for this problem?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
Send commands in Proxmark3 CLI mode.
First: ./pm3 -c 'hw ping'
Second: ./pm3 -c 'hw ping'
When I run the second command, I am unable to connect to the Proxmark hardware.
Afterwards, if you retry this command, the connection will be established normally.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows
    MCU....... AT91SAM7S512 Rev B
    Client.... Iceman/master/v4.17140-133-g9dac59a6b 2023-10-16 11:57:12
    Bootrom... Iceman/master/v4.16717-301-g03cc31cca-suspect 2023-08-06 14:35:40
    OS........ Iceman/master/v4.17140-139-g526632198-suspect 2023-10-16 19:50:18
    Target.... PM3 GENERIC

**Additional context**
Is there a delay set when reconnecting hardware? If not, can you please suggest me how to solve the problem?

iceman1001, 2023-10-19:
Can't replicate. 

- Check your usb cable? 
- Verify that your time is ok.    `hw ping; hw timeout`

[image]

dhje0ng, 2023-10-20:
In my case it was not a cable or hardware timeout related issue.

I tested both cases but could not resolve this issue.

I also tried reflashing the ICEMAN firmware on Proxmark3.

My Proxmark3 device is connected normally via USB cable and the LED is also active, but why doesn't the driver connect immediately?

You may need to wait 3 to 5 minutes for the device driver to be recognized.

My environment is building and running Proxmark3 ICEMAN in an Ubuntu 22.04.3 virtual environment.

iceman1001, 2023-10-20:
Which guide did you follow when setting up?   Have you removed the modem manager?

Your OS shouldn't take 3-5minutes to identify the device.  Something seems serious wrong.

dhje0ng, 2023-10-20:
To flash the ICEMAN version, we built it using the Linux build guide instructions below and then flashed the firmware using pm3-flash-all.

https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Linux-Installation-Instructions.md

https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/0_Compilation-Instructions.md

++)
After flashing the firmware, it will take 3-5 minutes for the device to reconnect to the Ubuntu VM.

I believe this is because the device must first be recognized by the host environment, Windows, so that it can make a choice about whether to connect this device to the VM.

I think the 3-5 minute wait is due to the fact that Windows cannot immediately recognize the proxmark3 device when it is connected.

dhje0ng, 2023-10-20:
Do you think there is a problem with the Proxmark3 equipment I have?
[image]

dhje0ng, 2023-10-20:
> Can't replicate.
> 
> * Check your usb cable?
> * Verify that your time is ok.    `hw ping; hw timeout`

> After reflashing the firmware of the Proxmark3 device, the issue we initially reported was resolved, but the device taking a long time to connect is a new issue.

iceman1001, 2023-10-20:
rdv2 should be fine.  

I see that you dídn't flash the bootrom... 
Make sure you flash it.

For Ubuntu VM,   I have no issue at all. I use it daily.

Which version Windows OS are you using?

dhje0ng, 2023-10-20:
You just flashed the bootrom area.
The windows I am using is Windows 10 PRO Build 22H2 19045.3570.

iceman1001, 2023-10-20:
Well,  if its not usb cable,   and its not usb hub  and you tested different usb ports,  and you have flashed latest source,
and the pm3 isn't damaged from something else.  (confirmed working with on a different system)

Then I would say it has nothing to do with the source code.    And more related to your OS/ WM USB send through. 

I suggest you ask questions on the discord server. 

closing.,

closed 2023-10-20 by iceman1001