RfidResearchGroup/proxmark3 #2158: [RELEASE 4.17421] Checklist - release name:  ???

iceman1001, 2023-11-05:
# Checklist

- [ ] CHANGELOG.md: add title: `## [releasename][YYYY-MM-DD]`
- [ ] `make style`
- [ ] `make miscchecks`
- [ ] `make clean; make client CC=clang CXX=clang++ LD=clang++` on recent Debian or Ubuntu
- [ ] `mymanualchecks.sh`
- [ ] `mycppcheck.sh` no alarming warning?
- [ ] `tools/build_all_firmwares.sh` check that the script contains all standalone modes then compile all standalone modes (linux only)
- [ ] `experimental_lib` compilation & tests
- [ ] `experimental_client_with_swig` compilation & tests
- [ ] GitHub Actions - green across the board ( MacOS, Ubuntu, Windows)

# OS compilation and tests

Run `tools/release_tests.sh` on:

- [ ] RPI Zero
- [ ] Jetson Nano
- [ ] WSL
- [ ] PSv3.10
- [ ] Archlinux
- [ ] Kali
- [ ] Debian Stable
- [ ] Debian Testing
- [ ] Ubuntu 22
- [ ] ParrotOS
- [ ] Fedora 37
- [ ] OpenSuse Leap
- [ ] OpenSuse Tumbleweed
- [ ] OSX (MacPorts)
- [ ] OSX (Homebrew)
- [ ] Android
- [ ] Termux

# creating release

- [ ] `make release RELEASE_NAME="ice awesome"`
  - last line of output,  gives you next command to run.
  - Sample:  `git push && git push origin v4.17421`
- [ ] CHANGELOG.md: edit title to add version info: `## [releasename.4.17421][YYYY-MM-DD]`

## Step Github releases

- [ ] Go to Github releases,  create release based on the new created tag and publish

## Step Homebrew updates

- [ ] update homebrew repo, file `proxmark3.rb`
  - with a SHA256 sum of the file `v4.17421.tar.gz`
  - with updated list of standalone modes

## Step package maintains

- [ ] make a list of new standalone modes, so when we alert package maintainers they have a sporting chance of adding them

doegox, 2023-11-13:
my tests ok, except experimental_client_with_swig still broken as for previous release https://github.com/RfidResearchGroup/proxmark3/issues/2104

doegox, 2023-11-13:
now experimental_client_with_swig fixed

doegox, 2023-11-13:
no new standalone modes

doegox, 2023-11-13:
https://github.com/RfidResearchGroup/proxmark3/releases/tag/v4.17511

closed 2023-11-13 by doegox