RfidResearchGroup/proxmark3 #2170: mfkey32 build error

JohnDoeIsKIng, 2023-11-13:
***Compilation problems***
Try compiling with verbose.  `make VERBOSE=1` with main makefile or `make V=1` with cmake.

Dear community,

with the actual code base I'm getting a build error:
`MAKE mfkey/all
[-] CC mfkey32.c
[-] CC ../../common/crapto1/crypto1.c
[-] CC ../../common/crapto1/crapto1.c
[-] CC ../../common/bucketsort.c
[-] CC nested_util.c
[=] CXX mfkey32
obj/nested_util.o: In Funktion »nested«:
nested_util.c:(.text+0x2b8): Warning: undefined reference »pthread_create«
nested_util.c:(.text+0x33f): Warning: undefined reference »pthread_join«
collect2: error: ld returned 1 exit status
../../Makefile.host:89: recipe for target 'mfkey32' failed
make[1]: *** [mfkey32] Error 1
Makefile:155: recipe for target 'mfkey/all' failed
make: *** [mfkey/all] Error 2
`

It came within the last 24 hours.
Any hints how to solve this ?
Best regards

JD.

phreakmonkey, 2023-11-14:
I get this issue too.

Platform:  Debian 11 (bullseye)

```
===================================================================
Version info:      Iceman/master/v4.17511-4-g2b320929f
Platform name:     Proxmark3 RDV4
PLATFORM:          PM3RDV4
PLATFORM_FPGA:     xc2s30
PLATFORM_SIZE:     512
Platform extras:   No extra selected
Included options:  SMARTCARD FLASH -DRDV4 LF HITAG EM4x50 EM4x70 ZX8211 GENERAL_HF ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   LF_SAMYRUN
===================================================================
[*] MAKE client/all
===================================================================
Version info:      Iceman/master/v4.17511-4-g2b320929f
Client platform:   Linux
GUI support:       QT5 found, enabled (Qt version 5.15.2 in /usr/lib/x86_64-linux-gnu)
native BT support: Bluez found, enabled
Jansson library:   system library not found, using local library
Lua library:       system library not found, using local library
Python3 library:   Python3 v3.9 found, enabled
Readline library:  enabled
Whereami library:  system library not found, using local library
Lua SWIG:          wrapper found
Python SWIG:       wrapper found
compiler version:  cc (Debian 10.2.1-6) 10.2.1 20210110
===================================================================
```
[...]
```
[=] CXX mfkey32
/usr/bin/ld: obj/nested_util.o: in function `nested':
nested_util.c:(.text+0x288): undefined reference to `pthread_create'
/usr/bin/ld: nested_util.c:(.text+0x30f): undefined reference to `pthread_join'
collect2: error: ld returned 1 exit status
make[1]: *** [../../Makefile.host:90: mfkey32] Error 1
make: *** [Makefile:156: mfkey/all] Error 2
```

It doesn't happen if I roll back to commit 5dd144b.  I'll walk the commit history and see where it was introduced.

phreakmonkey, 2023-11-14:
Found it in commit ecd7f53

Tested and created a PR with the preexisting test for pthreads.

https://github.com/RfidResearchGroup/proxmark3/pull/2176

phreakmonkey, 2023-11-15:
Oh look there's another PR just before mine that also addresses this.  :man_shrugging:

JohnDoeIsKIng, 2023-11-16:
@phreakmonkey :

Both PR work for me, thanx a lot!
Will remember this in similar situations (as i remember that wasn't the first time lpthread was needed extra).
Best regards

JD.

closed 2023-11-16 by JohnDoeIsKIng