RfidResearchGroup/proxmark3 #2180: No 'hw ping' or 'hw break' stop support for iclass simulation

00Waz, 2023-11-16:
The ability to perform 'hw ping' or 'hw break' commands in order to abort simulation of iclass cards is not supproted.
Last full release that i found that supports this is v4.9237, and the bug was introduced by v4.13441.
Emulation only stops by physical button press.

Example running latest fw.
hw ping failing to respond and stop emulation of mode 1 iclass.
<pre>
[usb] pm3 --> hw ping
[=] Ping sent
[+] Ping response received
[usb] pm3 --> hf iclass sim -t 1
[=] Starting iCLASS simulation
[=] press `button` to cancel
[usb] pm3 --> hw ping
[=] Ping sent
[!] ⚠️  Ping response timeout

[#] button pressed
[usb] pm3 --> hw ping
[=] Ping sent
[+] Ping response received
</pre>

example of mode 3 "full card simulation"
<pre>
[usb] pm3 --> hw ping
[=] Ping sent
[+] Ping response received
[usb] pm3 --> hf iclass eload -f ~/test_file.bin
[+] loaded 152 bytes from binary file `~/test_file.bin`

[=] Uploading to emulator memory
[=] ..
[+] uploaded 152 bytes to emulator memory
[?] You are ready to simulate. See `hf iclass sim -h`
[=] Done!
[usb] pm3 --> hf iclass sim -t 3
[=] Starting iCLASS simulation
[=] press `button` to cancel
[?] Try `hf iclass esave -h` to save the emulator memory to file
[usb] pm3 --> hw ping
[=] Ping sent
[!] ⚠️  Ping response timeout

[#] button pressed
[usb] pm3 --> hw ping
[=] Ping sent
[+] Ping response received
</pre>

wh201906, 2023-11-16:
I guess this is the reason:

In `v4.9237` (commit 833bc4d9a3f2c1dcef4a72393663f75e0ba0fbd1), the `doIClassSimulation()` calls `GetIClassCommandFromReader()` in the loop, which checks if the button is pressed and if there is any data from the client.
https://github.com/RfidResearchGroup/proxmark3/blob/833bc4d9a3f2c1dcef4a72393663f75e0ba0fbd1/armsrc/iclass.c#L1446
https://github.com/RfidResearchGroup/proxmark3/blob/833bc4d9a3f2c1dcef4a72393663f75e0ba0fbd1/armsrc/iclass.c#L1018

In `v4.13441` (commit 55e1b6714b7cccaf177285572d4c60db7d305255), the `do_iclass_simulation()` calls `GetIso15693CommandFromReader()` in the loop, which only checks if the button is pressed. (I guess `doIClassSimulation()` is renamed into `do_iclass_simulation()`)
https://github.com/RfidResearchGroup/proxmark3/blob/55e1b6714b7cccaf177285572d4c60db7d305255/armsrc/iclass.c#L516
https://github.com/RfidResearchGroup/proxmark3/blob/55e1b6714b7cccaf177285572d4c60db7d305255/armsrc/iso15693.c#L1165

iceman1001, 2023-11-17:
Simulation is a very time critical piece of code.  The check for data available distrupted it. 
There are some improvement checks in @wh201906  enhanced udp code,   not sure if its possible to squeese it down to fit here

wh201906, 2023-11-17:
I think the `data_available_fast()` added in PR #2173 can be used for this. It only checks few of the registers.

iceman1001, 2023-11-17:
yeah,     however ,  simulation fiddling means testing against real readers.  and iclass means testing against a bunch of different models,   since they are all a bit different in their time sensitivity

wh201906, 2023-11-17:
Indeed

closed 2024-05-13 by iceman1001