RfidResearchGroup/proxmark3 #2182: Building against Python 3.12 is broken

dogtopus, 2023-11-18:
**Describe the bug**

Python 3.12.0 has a bug (python/cpython#106560) that will cause the compilation of pm3 client to fail.

A workaround would simply be disabling `-Werror` for `redundant-decls`.

**To Reproduce**
Steps to reproduce the behavior:
1. Build proxmark3 client with Python 3.12.0
2. See it fail with `-Werror=redundant-decls`

**Expected behavior**
Build under Python 3.12.0 successfully.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Fedora 39
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
 - hw status
 - data tune

**Additional context**
Add any other context about the problem here.

wh201906, 2023-11-24:
Does the cmake work?

dogtopus, 2023-11-24:
@wh201906 Presumably no since it's a Python issue, unless CMake didn't honor `-Werror` by default for some reason.

wh201906, 2023-11-24:
I don't have Fedora 39 so I did a test roughly and it seems to only affect the `make` command. Could you please help me to test `cmake` on Fedora 39 with Python 3.12?
As for the `Makefile`, I've made PR #2185 to fix it.

iceman1001, 2023-11-24:
merged the fix.

closing

closed 2023-11-24 by iceman1001