RfidResearchGroup/proxmark3 #2197: Em410x emulation not working

Fonteno, 2023-12-06:
Reading and cloning works fine. But emulation is not working for em4100 cards. It says simulating but the reader is not picking anything.

piotrva, 2023-12-24:
What type of hardware do you have?
There are is lot of hardware on the market that is flawed unfortunately

iceman1001, 2023-12-24:
I just used the pm3 to simulate EM4100 against a Chameleon Ultra (CU).   The CU didn't have any issue reading the simulated tag. 

You will need to add details on how to replicate your issue.

piotrva, 2023-12-24:
I am simulating both 125kHz and 13.56 MHz using Proxmark3 RDV without any issue.
However there is Proxmark3 Easy:
1. I received one unit with a missing driver IC (so it was the reason why I was unable to run any simulation)
2. It was replaced by a seller and I got unit with all IC correctly populated. However only simulation of 125 KHz worked - 13.56 MHz was still not functioning.

So in this situation I assume @Fonteno might have similar hardware issue.

iceman1001, 2023-12-24:
OK,    
Thanks for clarifying.   

Closing since its a hardware issue.

closed 2023-12-24 by iceman1001

piotrva, 2024-12-26:
Also a note for future users - try running simulation with `--gap 0` argument - some readers expect a continuous, uninterrupted stream of data.

iceman1001, 2024-12-26:
Maybe we should improve the help text part. 
So we get two clear examples and a short explanation why.

piotrva, 2024-12-26:
I was thinking about it as well as making gap = 0 a default option. What do you think?