RfidResearchGroup/proxmark3 #2200: Add new keys to Dict

icemirr0r, 2023-12-09:
Hi, please add the following keys to mf_classic_dict.nfc

#Misc. keys from hotels & library cards in Germany
914f57280ce3
324a82200018
370aee95cd69
2e032ad6850d
1feda39d38ec
288b7a34dbf8
0965e3193497
18c628493f7f
064d9423938a
995fd2a2351e
7c7d672bc62e
217250fb7014
ae7478ccaee7
abbf6d116eaf
05862c58edfb
e43b7f185460
6a59aa9a959b
b79e5b175227
7bc9ebb8274b
b2afbf2331d4
223e5847dd79
640524d2a39b
aee297cb2fd6
3da5dfa54604
0cf1a2aa1f8d

iceman1001, 2023-12-09:
Nice,

Have you verified that these keys are static across multiple tags from the same system/hotel?

We only interested in static default keys.     The alternative is that keys are diversified ,  quite common with hospitality systems.

icemirr0r, 2023-12-09:
> Nice,
> 
> Have you verified that these keys are static across multiple tags from the same system/hotel?
> 
> We only interested in static default keys. The alternative is that keys are diversified , quite common with hospitality systems.

Yes, checked it with at least one additional card per location/facility.

iceman1001, 2023-12-09:
Awesome!

Ìf you remember,  Which systems are they from?

closed 2023-12-09 by iceman1001