RfidResearchGroup/proxmark3 #2213: Added read and write sniffing and copying of 15693 protocol slix2 class card

mayjack0312, 2023-12-25:
**Describe the solution you'd like**
I am trying to copy a card of 15693, which seems to use the latest encryption method slix2.
At first these tags looked perfect and worked like a charm. Later on after a bit of usage within the development process of my reader hardware, some features of the tag stoped working.
Some information about the NXP ICODE SLIX-L chip:
The SLIX-L tag is kind of special, because it has only 32 Byte of memory (8 Blocks with 4 Bytes per Block), an UID that is always starting with "E0 04 03..." and a privacy mode that once it is enabled to tag is not "visible" for any reader. With an enabled privacy mode the tag is just reacting to one custom command (random number) and keeps silent for all other standard commands like INVENTORY. With the random number and a preset password another custom command can be used to disable the privacy mode. After disabling this feature the tag is acting to al the standard commands like INVENTORY, READ or WRITE.
I figured out, that removing the tag from the reader rf field while sending the "enable privacy mode" command, lets the chip break which shows in a changing IC value of the chip from "03" to "01". That means after this special situation the chip is not behaving as a chip with an IC value of "03" which stands for the SLIX-L chip, but like a chip with an IC value of "01" which stands for an SLIX chip.
The SLIX chip does not support a privacy mode at all but has a memory size of 28 Blocks with 4 Bytes each compared to 8 Blocks of the SLIX-L chip.
That means the former tag with an SLIX-L chip is acting now as a tag with a SLIX chip including all specifications.
This chip has the ability to be whatever the seller wants it to be by changing the IC value of the chip. In addition there must be the possibility of changing the UID as well. Because within the specification it says, that the first three Bytes of a SLIX-L UID are "E0 04 03..." and the first three Byte of a SLIX UID are "E0 04 01...". This change has do be done in an own step, because the UID is still the same like before. Therefore must be a change feature for the UID available.(How to crack)
The accidental change of the IC value must be a bug. But now I am curious to know how to reset the IC value to "03" or get to know how to change the UID.How to crack
I looked it up: To disable privacy mode, you can use Proxmark3 or use the "knock method". But I found that the "knock method" may not suit me, but by looking up [project documentation](https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/commands.md) I also found out how to use it PM3 to unblock:
hf 15 slixprivacydisable | N | Disable privacy mode on SLIX ISO-15693 tag
-- | -- | --
hf 15 slixprivacyenable | N | Enable privacy mode on SLIX ISO-15693 tag

  
But I still encountered the above problem when copying, which caused the card reader to not recognize the card at all after copying. Please help.
The gate I use is AX500 Smart Gate NG - Flap

**Additional context**
[image]
[image]
[image]
[image]
[image]
[image]

iceman1001, 2024-01-26:
We have normal dump / restore commands and a new file format that handles variable block sizes.
You should be good with that to start with. 
for extras you need to figure out what the exact process is and reproduce the steps using the pm3.

I don't have a SLIX-L  to test with and its up to you now. 

We welcome pull requests!

iceman1001, 2025-06-08:
Ping,   since we have few active maintainers its comes down to you.   Make it happen!

Meanwhile I am closing this

hmdevksk, 2025-11-21:
Hello, Did you solve it? I'm experiencing the same issue. I enabled privacy mode (hf 15 slixprivacyenable) on a 40-block SLIX-S tag (with a chip IC value of 02), and it changed to 28 blocks and the chip IC value to 01, but I can't do hf 15 slixprivacydisable again.