RfidResearchGroup/proxmark3 #2221: HF antenna is UNUSABLE

cletifer, 2023-12-29:
I just bought a brand new out of the box Proxmark3 RDV4.01 with Blueshark from Hacker Warehouse while folding the Guide on this github page and running the hw tune command I noticed the HF antenna is unusable and am looking for some help troubleshooting this. I have double-checked the screws on the antenna as well.  

Desktop (please complete the following information):
cat /etc/os-release 
PRETTY_NAME="Ubuntu 22.04.3 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.3 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy

[usb] pm3 --> hw status 
[#] Memory
[#]   BigBuf_size............. 38460
[#]   Available memory........ 36152
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... fpga_pm3_hf.ncd image 2s30vq100 2023-08-29 16:44:19
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... OK
[#]   Memory size............. 2 mbits / 256 kb
[#]   Unique ID (be).......... 0x39163E53175462D5
[#] Smart card module (ISO 7816)
[#]   version................. v4.42 ( ok )
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
[#]     long leading reference |  29 |  17 |  18 |  50 |  15 | N/A | N/A | 
[#]               leading zero |  29 |  17 |  18 |  40 |  15 | N/A | N/A | 
[#]    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 336896
[#]   Transfer Speed PM3 -> Client... 673792 bytes/s
[#] Various
[#]   Max stack usage......... 3520 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... -1
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 31581 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[#] Flash memory dictionary loaded
[#]   Mifare.................. 1718 / 2047 keys
[#]   T55x7................... 123 / 1023 keys
[#]   iClass.................. 28 / 511 keys
[#]
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ Client ]
  Iceman/master/v4.17511-201-g56c318785-suspect 2023-12-28 17:03:30 b48accca8
  compiled with............. GCC 11.4.0
  platform.................. Linux / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... present
  Python script support..... present
  Lua SWIG support.......... present
  Python SWIG support....... present

 [ Proxmark3 ]
  device.................... RDV4
  firmware.................. RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... present

 [ ARM ]
  bootrom: Iceman/master/release (git)-suspect  b48accca8
       os: Iceman/master/v4.17511-201-g56c318785-suspect 2023-12-28 19:02:03 b48accca8
  compiled with GCC 10.3.1 20210621 (release)

 [ FPGA ] 
  fpga_pm3_lf.ncd image 2s30vq100 2023-08-29 16:44:07
  fpga_pm3_hf.ncd image 2s30vq100 2023-08-29 16:44:19
  fpga_pm3_felica.ncd image 2s30vq100 2023-08-29 16:44:43
  fpga_pm3_hf_15.ncd image 2s30vq100 2023-08-29 16:44:31

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 69% used)
[usb] pm3 --> hw tune
[=] ---------- Reminder ------------------------
[=] `hw tune` doesn't actively tune your antennas,
[=] it's only informative.
[=] Measuring antenna characteristics, please wait...
 🕙   9
[=] ---------- LF Antenna ----------
[+] LF antenna: 73.62 V - 125.00 kHz
[+] LF antenna: 32.05 V - 134.83 kHz
[+] LF optimal: 73.62 V - 125.00 kHz
[+] Approx. Q factor (*): 12.4 by frequency bandwidth measurement
[+] Approx. Q factor (*): 12.8 by peak voltage measurement
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[!] ⚠️  HF antenna is UNUSABLE

(*) Q factor must be measured without tag on the antenna

[+] Displaying LF tuning graph. Divisor 88 (blue) is 134.83 kHz, 95 (red) is 125.00 kHz.

pickeditmate, 2023-12-29:
You're looking for 
hf tune

iceman1001, 2023-12-29:
Issues here are solely about source code.  Your issue seems to be related to HW.
Join the discord server and ask there.

Meanwhile,   flash bootrom aswell.

closed 2023-12-29 by iceman1001

cletifer, 2023-12-29:
@iceman1001 can you provide me the link to get to the discord?

iceman1001, 2023-12-30:
Its in the readme of this repository.

cletifer, 2023-12-30:
Thank you very much.