RfidResearchGroup/proxmark3 #2228: [RELEASE 4.17764] Checklist - release name Steamboat Willie

iceman1001, 2024-01-02:
# Checklist

- [ ] `make style`
- [ ] `make miscchecks`
- [ ] `make clean; make client CC=clang CXX=clang++ LD=clang++` on recent Debian or Ubuntu
- [ ] `mymanualchecks.sh`
- [ ] `mycppcheck.sh` no alarming warning?
- [ ] `tools/build_all_firmwares.sh` check that the script contains all standalone modes then compile all standalone modes (linux only)
- [ ] `experimental_lib` compilation & tests
- [ ] `experimental_client_with_swig` compilation & tests
- [ ] GitHub Actions - green across the board ( MacOS, Ubuntu, Windows)

# OS compilation and tests

Run `tools/release_tests.sh` on:

- [ ] RPI Zero
- [ ] Jetson Nano
- [ ] WSL
- [ ] PSv3.10
- [ ] Archlinux
- [ ] Kali
- [ ] Debian Stable
- [ ] Debian Testing
- [ ] Ubuntu 22
- [ ] ParrotOS
- [ ] Fedora 37
- [ ] OpenSuse Leap
- [ ] OpenSuse Tumbleweed
- [ ] OSX (MacPorts)
- [ ] OSX (Homebrew)
- [ ] Android
- [ ] Termux

# creating release

- [ ] CHANGELOG.md: add title: `## [Steamboat Willie][YYYY-MM-DD]`
- [ ] `make release RELEASE_NAME="Steamboat Willie"`
  - last line of output gives you next command to run.
  - Sample:  `git push && git push origin v4.17764`
- [ ] CHANGELOG.md: edit title to add version info: `## [Steamboat Willie.4.17764][YYYY-MM-DD]`

## Step Github releases

- [ ] Go to Github releases,  create release based on the new created tag and publish
  - Choose a tag: v4.17764
  - Target: master
  - Set as the latest release
  - Title: `proxmark3-v4.17764`
  - Description:
```
Release v4.17764
Nickname "Steamboat Willie"
```

## Step Homebrew updates

- [ ] update homebrew repo, file `proxmark3.rb`
  - with a SHA256 sum of the file `v4.17764.tar.gz`
  - with updated list of standalone modes

## Step package maintains

- [ ] make a list of new standalone modes, so when we alert package maintainers they have a sporting chance of adding them

iceman1001, 2024-01-02:
New standalone mode in this release:  `hf_unisniff`  

And from the previous release we also had `hf_cardhopper`

iceman1001, 2024-01-02:
@doegox  are you able to run the docker environment tests?

closed 2024-01-03 by iceman1001

doegox, 2024-01-03:
visibly you managed it :)