RfidResearchGroup/proxmark3 #2235: Compilation Errors / MAC M1 / error: target attribute is not supported on this machine

cypher2001, 2024-01-05:
In process of creating fresh files on new release. Previous versions compiled just fine on system.

MAC M1 / Sonoma 14.2.1

make VERBOSE=1 clean && make VERBOSE=1 -j 

Finishes but pops 3 errors.

[-] CXX sma_multi.cpp
[-] CC ../common_arm/ticks.c
In file included from ../common_arm/ticks.c:19:0:
../common_arm/ticks.h:34:1: error: target attribute is not supported on this machine [-Werror=attributes]
 uint32_t RAMFUNC GetCountUS(void);
 ^
In file included from ../common_arm/flashmem.c:22:0:
../common_arm/ticks.h:34:1: error: target attribute is not supported on this machine [-Werror=attributes]
 uint32_t RAMFUNC GetCountUS(void);
 ^
../common_arm/ticks.c:253:1: error: target attribute is not supported on this machine [-Werror=attributes]
 uint32_t RAMFUNC GetCountUS(void) {
 ^
[-] CC version_pm3.c

iceman1001, 2024-01-05:
Thanks for reporting, 
If you are able to find a fix  then that would be nice.

Something tells me that your compiler doesn't like the `#define RAMFUNC`   .    That where I would have started.

GhostDog98, 2024-09-04:
I'm still facing this issue which makes compiling on my mac impossible. Did you end up finding a fix @cypher2001 ? I'm on an i5 macbook pro from 2020...

GhostDog98, 2024-09-04:
Update: After running the following:
```
brew unlink arm-none-eabi-gcc
brew install RfidResearchGroup/proxmark3/arm-none-eabi-gcc
```
It worked. 
My working theory is that macos installs a very old version of this package with xcode or something else, and this is causing the above error.

iceman1001, 2024-09-09:
closing since the solution is outside of this repo

closed 2024-09-09 by iceman1001