RfidResearchGroup/proxmark3 #2236: hf_legic_clone script do not run //  attempt to index local 'readbytes' (a nil value)

thebasti314, 2024-01-05:
Hello!

I try to clone a Legic Prime Card to another card using the hf_legic_clone Script.

Unfortunately the script returns an error.

Any Ideas?

Have good day!

`[usb] pm3 --> script run hf_legic_clone -i /Users/acdf/RFID/01_new/hf-legic-7111C08C-dump.bin -w
[+] executing lua /opt/homebrew/Cellar/proxmark3/4.17768/bin/../share/proxmark3/luascripts/hf_legic_clone.lua
[+] args '-i /Users/acdf/RFID/01_new/hf-legic-7111C08C-dump.bin -w'

read 1024 bytes from /Users/acdf/RFID/01_new/hf-legic-7111C08C-dump.bin

place your empty tag onto the PM3 to restore the data of the input file
the CRCs will be calculated as needed
 confirm when ready [y/n] ?y

ERROR:	Error, waiting for response timed out :: No response from the device

these are the MCD MSN0 MSN1 MSN2 from the Tag that has being read:
[-] ⛔ error - ...768/bin/../share/proxmark3/luascripts/hf_legic_clone.lua:413: attempt to index local 'readbytes' (a nil value)`

 OS: MacOS 14.21

`[usb] pm3 --> hw version
`
[ Proxmark3 RFID instrument ]

[ Client ]
Iceman/master/v4.17768-suspect 2024-01-03 01:59:39 4aa4f5078
compiled with............. Clang/LLVM Apple LLVM 15.0.0 (clang-1500.1.0.2.5)
platform.................. OSX / aarch64
Readline support.......... present
QT GUI support............ present
native BT support......... absent
Python script support..... present
Lua SWIG support.......... present
Python SWIG support....... present

[ Proxmark3 ]
firmware.................. PM3 GENERIC

[ ARM ]
bootrom: Iceman/master/v4.17768-suspect 2024-01-03 01:59:39 4aa4f5078
os: Iceman/master/v4.17768-suspect 2024-01-03 01:59:39 4aa4f5078
compiled with GCC 10.3.1 20210824 (release)

[ FPGA ]
fpga_pm3_lf.ncd image 2s30vq100 2023-08-29 16:44:07
fpga_pm3_hf.ncd image 2s30vq100 2023-08-29 16:44:19
fpga_pm3_felica.ncd image 2s30vq100 2023-08-29 16:44:43
fpga_pm3_hf_15.ncd image 2s30vq100 2023-08-29 16:44:31

[ Hardware ]
--= uC: AT91SAM7S512 Rev A
--= Embedded Processor: ARM7TDMI
--= Internal SRAM size: 64K bytes
--= Architecture identifier: AT91SAM7Sxx Series
--= Embedded flash memory 512K bytes ( 61% used ) `

iceman1001, 2024-01-05:
the biggest issue with legic for us is that noone has access to the different dumps / reader that is needed to verify the functionality of the script.

So you are the best person to do it!

thebasti314, 2024-01-05:
Okay. What do I have to do? ☺️

iceman1001, 2024-01-05:
debug the script ....

PR #2393 (2024-05-25): fix: hf_legic_clone.lua script
Fixes: #2236
commit 4babe8f0 (2024-05-25): fix: hf_legic_clone.lua script

Fixes: #2236
closed 2024-05-25 by iceman1001