RfidResearchGroup/proxmark3 #2241: Hitag2 write block 0 feature

csjanos, 2024-01-08:
In the case of certain manufactured hitag2 cards, the 0 block can be written, but this is not currently implemented by proxmark.
It would be nice if there could be a function extension that would check whether the zero block can be written and if so, it would enable its writing.
It would also be useful if the hitag info also states whether the zero block can be written, because it does not provide information about this right now.

iceman1001, 2024-01-08:
Which certain manufactured hitag2 cards is that?

csjanos, 2024-01-09:
It is like Super Chip XT27A01 XT27A66 Transponders, PCF7936 chip type

csjanos, 2024-01-22:
If I can help you in any way in terms of development/testing, feel free to let me know, because I'd be happy to.

iceman1001, 2024-05-13:
The XT27A01  type of transponders uses their own proprietary command set.   
You would need to reverse it in order to get support for in the PM3.

closed 2024-05-13 by iceman1001