RfidResearchGroup/proxmark3 #2243: Legic Simulation seem to break midway

tsteinruecken, 2024-01-12:
I have issues simulating a Legic Prime Tag. Dumping and writing tags seem to work fine (the data in my dump is highly plausible), but i get no response from the reader when simulating. From the trace it looks like the first messages are correctly handled, but then the simulation freezes and does not respond to further commands.

[image]

[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ Client ]
  Iceman/master/v4.17768-74-g261760e90-dirty-suspect 2024-01-12 15:20:21 a3de9f912
  compiled with............. GCC 13.2.1 20231205 (Red Hat 13.2.1-6)
  platform.................. Linux / x86_64
  Readline support.......... present
  QT GUI support............ absent
  native BT support......... absent
  Python script support..... absent
  Lua SWIG support.......... present
  Python SWIG support....... absent

 [ Proxmark3 ]
  device.................... device / fw mismatch
  firmware.................. RDV4
  external flash............ present
  smartcard reader.......... absent
  FPC USART for BT add-on... absent

 [ ARM ]
  bootrom: Iceman/master/v4.17768-74-g261760e90-dirty-suspect 2024-01-12 14:33:31 b8762748b
       os: Iceman/master/v4.17768-74-g261760e90-dirty-suspect 2024-01-12 15:21:13 a3de9f912
  compiled with GCC 13.1.0

 [ FPGA ] 
  fpga_pm3_lf.ncd image 2s30vq100 2023-08-29 16:44:07
  fpga_pm3_hf.ncd image 2s30vq100 2023-08-29 16:44:19
  fpga_pm3_felica.ncd image 2s30vq100 2023-08-29 16:44:43
  fpga_pm3_hf_15.ncd image 2s30vq100 2023-08-29 16:44:31

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 68% used )

[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 40548
[#]   Available memory........ 40548
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... fpga_pm3_hf.ncd image 2s30vq100 2023-08-29 16:44:19
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... ok
[#]   Unique ID (be).......... 0x2439BA4B64B062DC
[#] Smart card module (ISO 7816)
[#]   version................. ( fail )
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
[#]     long leading reference |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
[#]               leading zero |  29 |  17 |  15 |  40 |  15 | N/A | N/A | 
[#]    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 377856
[#]   Transfer Speed PM3 -> Client... 755712 bytes/s
[#] Various
[#]   Max stack usage......... 3520 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... -1
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 30092 Hz
[#] Installed StandAlone Mode
[#]   HF Legic Prime standalone
[#] Flash memory dictionary loaded

[usb] pm3 --> hw tune
[=] ---------- Reminder ------------------------
[=] `hw tune` doesn't actively tune your antennas,
[=] it's only informative.
[=] Measuring antenna characteristics, please wait...
 🕛   9
[=] ---------- LF Antenna ----------
[+] At 125.00 kHz .......... 22.11 V
[+] At 134.83 kHz .......... 14.66 V
[+] At 117.65 kHz optimal... 25.25 V
[+] 
[+] Approx. Q factor measurement (*)
[+] Frequency bandwidth..... 6.3
[+] Peak voltage............ 4.4
[!] ⚠  Contradicting measures seem to indicate you're running a PM3_RDV4 firmware on a generic device
[!] ⚠  False positives is possible but please check your setup
[+] LF antenna ( ok )

[=] ---------- HF Antenna ----------
[+] 13.56 MHz............... 39.87 V
[+] 
[+] Approx. Q factor measurement (*)
[+] peak voltage............ 7.0
[+] HF antenna ( ok )

(*) Q factor must be measured without tag on the antenna

[=] -------- LF tuning graph ---------

[+] Blue line  Divisor 88 / 134.83 kHz
[+] Red line   Divisor 95 / 125.00 kHz

No GUI in this build!

iceman1001, 2024-01-13:
Its a known issue and we can't debug it against a genuine reader,  

Its down to you to find the bug and apply a fix.

iceman1001, 2025-06-08:
Ping,    I figure you didn't find a genuine reader to help you fix the code for it.   Closing meanwhile

closed 2025-06-08 by iceman1001