RfidResearchGroup/proxmark3 #2244: hf 15 cmd -k option (keep field on) is broken/removed

Chris-P-Young, 2024-01-13:
***Description***
The hf 15 cmd has the option to keep RF field on between sending raw commands. This is the -k option.
At some point, this feature has been removed/deliberately broken for some reason.
The option is still present in the command help and is still supported as an option by the command parser.
However, the DirectTag15693Command in iso15693.c now explicitly turns off the field at the end of the command. It also unconditionally resets the field at the start of the command.

***Why is this important?***
Raw mode is useful to explore tag behaviour and custom commands.
Certain tag commands need to remain powered between commands. This means the RF field should stay on.
An example is the NXP ICODE SLIX-L which requires that the tag remains powered during the privacy disable sequence (get random number followed by set password command).
We already have instruction support for SLIX privacy enable/disable, but we should also be able to achieve the same thing using raw commands as Proxmark is meant to be a learning and investigative tool.

Verified with NXP ICODE SLIX-L tag and RRG Field detector fob.
 
***To reproduce***
Using NXP ICODE SLIX-L tag as an example.

Issue raw command GET RANDOM NUMBER, keep field on
hf 15 raw -k -c -2 -d 02b204

Tag will respond with 5 octets which includes the random number.

Issue raw command SET PASSWORD (for privacy password)
hf 15 raw -c -2 -d 02B30404xxxxxxxx (where xxxxxxxx is XOR'D password)

Client will return 
! command failed

The -k option is processed by the client but ignored. The field is removed between raw commands which resets the tag and invalidates the random number. Therefore the set password command will always fail.

***Proposed Solution***
A PR has already been filed (PR #1636) to add a fix to handle the -k option properly.
However, this PR is based on code using the older return code method and has not been picked up for a merge.
I will update and file a new PR.

A second issue is the handling of the timing for Iso15693 raw commands.
The DirectTag15693Command contains some code that parses the raw command string passed to it and uses this to determine whether the command should be transmitted with reader timing or "write alike" timing which has a much longer timeout.
For commands not recognised by this code, the standard (shorter) reader timeout is used which will cause the tag reply to be missed.
Currently this is the case when sending the SLIX-L "set password" and "enable privacy" commands. Both of these commands have "write alike" timing and require the longer timeout.
Without this,  the tag response is missed and PM3 client returns "command failed"

A nasty fix is to add the two extra commands to the switch/case structure to ensure that write alike timing is used.

    switch (data[1]) {
        case ISO15693_SET_PASSWORD:
        case ISO15693_ENABLE_PRIVACY:

This is the method that is used in the pull request I propose for the main issue.
However, this code should be re-written to address this is in a better way. The dependency on the command that is passed needs to be removed as this is meant to be "raw mode" and not tied to any specific command.
An extra command option could be added to the hf 15 raw command to specify write alike timing. Default timing would still be as reader.

I will propose this in a separate issue report.

iceman1001, 2024-01-13:
I agree,   the PR  and the current 15 raw code needs some love with being converted to use a struct for all values.

You find my comments in the PR.

iceman1001, 2024-01-14:
try https://github.com/RfidResearchGroup/proxmark3/commit/8d0b41a911ca308eac990cc6bdb21d3fd073ef1c  and see if this works better.

commit 8d0b41a9 (2024-01-14): this commit fixes #2244 #2246 #1596 #2101.  Its kind of a big refactoring and I most likely broke something. With that said.  Now:  HF 15 commands now uses NG packets,  hf 15 raw support -k keepfield on and -s select,   hf 15 dump/rdbl/rdmulti should handle blocksizes of 4 or 8,  the error messages are unified and error handling the same.  Some understanding how add_option impacts response message from card. A more clear separation between PM3 flags and ISO15693 protocol flags.
closed 2024-01-14 by iceman1001