RfidResearchGroup/proxmark3 #2248: Issues with EM4425 : anyone tried that chip ?

BorisDess, 2024-01-14:
Hello everyone.

I have 3 EM Echo V samples at my disposal (ISO15693).
However, only one unit (tag A) can be read, but not written with a Proxmark.

On the other hand three units can be read and memory dumped with a smartphone (NXP apps or RFID Tools)
Only one can be written (tag B).

It's very strange to see such incoherence when they all should be the same chip.

These units are from different Tags manufacturers.
Does anyone has experience with this EM Echo V chip ?
I'm guessing they aren't broadly adopted and it's a matter of EM badly implementing the ISO standard ?

iceman1001, 2024-01-14:
Have you run this firmware on your device when testing?

BorisDess, 2024-01-14:
I followed : 
`https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/macOS-Homebrew-Installation-Instructions.md#macos---homebrew-automatic-installation`
And ran : 
`pm3-flash-all
`
And I get : 

    MCU....... AT91SAM7S512 Rev B
    Memory.... 512 KB ( 61% used )
    Client.... Iceman/master/v4.17768-76-g224da75e7 2024-01-14 13:12:14
    Bootrom... Iceman/master/v4.17768-76-g224da75e7-suspect 2024-01-14 13:12:16 
    OS........ Iceman/master/v4.17768-76-g224da75e7-suspect 2024-01-14 13:12:16 
    Target.... PM3 GENERIC

Am I good ?

iceman1001, 2024-01-14:
Is the the source code issues,   
For user related questions please use the  discord server.    It is its purpose.

closed 2024-01-14 by BorisDess