RfidResearchGroup/proxmark3 #2259: proxmark3 easy does not appear in bootloader mode

kyloren20001, 2024-01-19:
I'm using the VM Parrot OS with Proxmark3 Easy.
Follow inctructions to the letter, but i do not get the result.
```
┌─[root@kyloren-virtualbox]─[/commune/proxmark3-master/proxmark3-master]
└──╼ #sudo dmesg | grep -i usb
[    1.419159] ACPI: bus type USB registered
[    1.419175] usbcore: registered new interface driver usbfs
[    1.419180] usbcore: registered new interface driver hub
[    1.419187] usbcore: registered new device driver usb
[    1.437751] ohci-pci 0000:00:06.0: new USB bus registered, assigned bus number 1
[    1.501021] usb usb1: New USB device found, idVendor=1d6b, idProduct=0001, bcdDevice= 6.01
[    1.501026] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.501027] usb usb1: Product: OHCI PCI host controller
[    1.501028] usb usb1: Manufacturer: Linux 6.1.0-1parrot1-amd64 ohci_hcd
[    1.501029] usb usb1: SerialNumber: 0000:00:06.0
[    1.501217] hub 1-0:1.0: USB hub found
[   91.633457] usb 1-1: new full-speed USB device number 2 using ohci-pci
[   92.184262] usb 1-1: New USB device found, idVendor=9ac4, idProduct=4b8f, bcdDevice= 0.01
[   92.184279] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   92.184281] usb 1-1: Product: proxmark3
[   92.184282] usb 1-1: Manufacturer: proxmark.org
[   92.316769] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[   92.316797] usbcore: registered new interface driver cdc_acm
[   92.316798] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[  173.213056] usb 1-1: USB disconnect, device number 2
[  233.254569] usb 1-1: new full-speed USB device number 3 using ohci-pci
[  233.800740] usb 1-1: New USB device found, idVendor=9ac4, idProduct=4b8f, bcdDevice= 0.01
[  233.800745] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  233.800746] usb 1-1: Product: proxmark3
[  233.800747] usb 1-1: Manufacturer: proxmark.org
[  233.814879] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[  341.997063] usb 1-1: USB disconnect, device number 3
[  372.620601] usb 1-1: new full-speed USB device number 4 using ohci-pci
[  373.152823] usb 1-1: New USB device found, idVendor=9ac4, idProduct=4b8f, bcdDevice= 0.01
[  373.152828] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  373.152829] usb 1-1: Product: proxmark3
[  373.152831] usb 1-1: Manufacturer: proxmark.org
[  373.166634] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[  437.972166] usb 1-1: USB disconnect, device number 4
[  502.887529] usb 1-1: new full-speed USB device number 5 using ohci-pci
[  503.434662] usb 1-1: New USB device found, idVendor=9ac4, idProduct=4b8f, bcdDevice= 0.01
[  503.434666] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  503.434668] usb 1-1: Product: proxmark3
[  503.434669] usb 1-1: Manufacturer: proxmark.org
[  503.449450] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[  868.622157] usb 1-1: USB disconnect, device number 5
[  878.466929] usb 1-1: new full-speed USB device number 6 using ohci-pci
[  879.012925] usb 1-1: New USB device found, idVendor=9ac4, idProduct=4b8f, bcdDevice= 0.01
[  879.012930] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  879.012931] usb 1-1: Product: proxmark3
[  879.012932] usb 1-1: Manufacturer: proxmark.org
[  879.029150] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[  888.989261] usb 1-1: USB disconnect, device number 6
[  981.202735] usb 1-1: new full-speed USB device number 7 using ohci-pci
[  981.750832] usb 1-1: New USB device found, idVendor=9ac4, idProduct=4b8f, bcdDevice= 0.01
[  981.750837] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  981.750839] usb 1-1: Product: proxmark3
[  981.750840] usb 1-1: Manufacturer: proxmark.org
[  981.762872] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[  999.158238] usb 1-1: USB disconnect, device number 7
[ 1368.598723] usb 1-1: new full-speed USB device number 8 using ohci-pci
[ 1369.151858] usb 1-1: New USB device found, idVendor=9ac4, idProduct=4b8f, bcdDevice= 0.01
[ 1369.151863] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1369.151864] usb 1-1: Product: proxmark3
[ 1369.151865] usb 1-1: Manufacturer: proxmark.org
[ 1369.169078] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
┌─[root@kyloren-virtualbox]─[/commune/proxmark3-master/proxmark3-master]
└──╼ #./pm3 --list
1: /dev/ttyACM0
```
When i'm running the 'pm3-flash-all' command gives the following:
```
./pm3-flash-all
[=] Session log /root/.proxmark3/logs/log_20240119075640.txt
[+] About to use the following files:
[+]    /commune/proxmark3-master/proxmark3-master/client/../bootrom/obj/bootrom.elf
[+]    /commune/proxmark3-master/proxmark3-master/client/../armsrc/obj/fullimage.elf
[+] Loading ELF file /commune/proxmark3-master/proxmark3-master/client/../bootrom/obj/bootrom.elf
[+] ELF file version Iceman/master/release (git)-suspect  6233e2116

[+] Loading ELF file /commune/proxmark3-master/proxmark3-master/client/../armsrc/obj/fullimage.elf
[+] ELF file version Iceman/master/release (git)-suspect  6233e2116

[+] Waiting for Proxmark3 to appear on /dev/ttyACM0
 🕑  59 found
[+] Entering bootloader...
```
the pm3 s now shining with two signals, green snd orange but:
```
[+] (Press and release the button only to abort)
[+] Waiting for Proxmark3 to appear on /dev/ttyACM0
 🕒   0
[!] ⚠️  ERROR: invalid serial port /dev/ttyACM0

[?] Try the shell script `./pm3 --list` to get a list of possible serial ports
[!!] 🚨 Error: Proxmark3 not found.
[!] ⚠️  The flashing procedure failed, follow the suggested steps!
[!] ⚠️  Sending bytes to Proxmark3 failed ( offline )
[+] All done
```
Makefile.platform:
```
# If you want to use it, copy this file as Makefile.platform and adjust it to your needs
# Run 'make PLATFORM=' to get an exhaustive list of possible parameters for this file.

#PLATFORM=PM3RDV4
PLATFORM=PM3GENERIC
# If you want more than one PLATFORM_EXTRAS option, separate them by spaces:
#PLATFORM_EXTRAS=BTADDON
#PLATFORM_EXTRAS=FLASH
#PLATFORM_EXTRAS=SMARTCARD
#PLATFORM_EXTRAS=BTADDON FLASH
#STANDALONE=LF_SAMYRUN

# Uncomment the lines below in order to make a 256KB image
# and comment out the lines above

#PLATFORM=PM3GENERIC
#PLATFORM_SIZE=256
#STANDALONE=
#SKIP_HITAG=1
#SKIP_FELICA=1
#SKIP_HFPLOT=1
#SKIP_NFCBARCODE=1
#SKIP_ZX8211=1
#SKIP_LF=1

# To accelerate repetitive compilations:
# Install package "ccache" -> Debian/Ubuntu: /usr/lib/ccache, Fedora/CentOS/RHEL: /usr/lib64/ccache
# And uncomment the following line
#export PATH := /usr/lib64/ccache:/usr/lib/ccache:${PATH}

# To install with sudo:
INSTALLSUDO=sudo
```

iceman1001, 2024-01-19:
Which guide did you follow when installing?

It seem to me you skipped the modem manager / user access rights parts...

kyloren20001, 2024-01-21:
Actually I've followed this one
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Linux-Installation-Instructions.md

Then, i used the compilation one 
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/0_Compilation-Instructions.md

Oh, i see, i had skipped this one you've mentioned

kyloren20001, 2024-01-24:
i've deleted modemmanager, but it did nothing and the proxmark is still invisible when it enters bootloader
```
└──╼ #./pm3-flash-all
[=] Session log /root/.proxmark3/logs/log_20240124075916.txt
[+] About to use the following files:
[+]    /commune/proxmark3-master/proxmark3-master/client/../bootrom/obj/bootrom.elf
[+]    /commune/proxmark3-master/proxmark3-master/client/../armsrc/obj/fullimage.elf
[+] Loading ELF file /commune/proxmark3-master/proxmark3-master/client/../bootrom/obj/bootrom.elf
[+] ELF file version Iceman/master/release (git)-suspect  6233e2116

[+] Loading ELF file /commune/proxmark3-master/proxmark3-master/client/../armsrc/obj/fullimage.elf
[+] ELF file version Iceman/master/release (git)-suspect  6233e2116

[+] Waiting for Proxmark3 to appear on /dev/ttyACM0
 🕑  59 found
[+] Entering bootloader...
[+] (Press and release the button only to abort)
[+] Waiting for Proxmark3 to appear on /dev/ttyACM0
 🕒   0
[!] ⚠️  ERROR: invalid serial port /dev/ttyACM0

[?] Try the shell script `./pm3 --list` to get a list of possible serial ports
[!!] 🚨 Error: Proxmark3 not found.
[!] ⚠️  The flashing procedure failed, follow the suggested steps!
[!] ⚠️  Sending bytes to Proxmark3 failed ( offline )
[+] All done

[=] Have a nice day!

```

iceman1001, 2024-01-24:
ok,  this is no longer a source code issue,  but a user related issue.   For those we have the discord server.  

Closing.

closed 2024-01-24 by iceman1001