RfidResearchGroup/proxmark3 #2268: hf 15 writedsfid command doesn`t work

w3a4gh69, 2024-01-24:
[ Proxmark3 RFID instrument ]

    MCU....... AT91SAM7S512 Rev B
    Memory.... 512 KB ( 61% used )

    Client.... Iceman/master/v4.17768-204-g2d34713e5 2024-01-24 08:26:10
    Bootrom... Iceman/master/v4.17768-204-g2d34713e5-suspect 2024-01-24 08:25:48
    OS........ Iceman/master/v4.17768-204-g2d34713e5-suspect 2024-01-24 08:25:59
    Target.... PM3 GENERIC

hf 15 writedsfid command doesn`t work properly.
ex) hf 15 writedsfid --dsfid 0x02

can change nothing

iceman1001, 2024-01-24:
what is the output of 
```
hf 15 writedsfid --dsfid 0x02
hf 15 list
```

w3a4gh69, 2024-01-24:
[=] --- Tag Information ---------------------------
[+]     TYPE... EM-Marin SA (Skidata); EM4237 [IC id = 15] 23,5pF
[+]      UID... E0 16 3C 66 19 52 48 EF
[+]  SYSINFO... 00 0F EF 48 52 19 66 3C 16 E0 00 00 1B 03 01
[+]      - DSFID supported        [0x00]
[+]      - AFI   supported        [0x00]
[+]      - IC reference supported [0x01]
[+]      - Tag provides info on memory layout (vendor dependent)
[+]            4 (or 3) bytes/blocks x 28 blocks

[usb] pm3 --> hf 15 writedsfid --dsfid 0x02

[+] Wrote DSFID 0x02 ( ok )
[usb] pm3 --> hf 15 list
[+] Recorded activity ( 135 bytes )
[=] start = start of start frame. end = end of frame. src = source of
transfer.
[=] ISO15693 / iCLASS - all times are in carrier periods (1/13.56MHz)

      Start |        End | Src | Data (! denotes parity error)
                              | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |      22016 | Rdr |26  01  00  f6  0a
                              |  ok | INVENTORY
          0 |      22016 | Rdr |26  01  00  f6  0a
                              |  ok | INVENTORY
      26304 |      79552 | Tag |00  00  ef  48  52  19  66  3c  16  e0  ec
 a9                           |  ok |
        256 |      50944 | Rdr |22  2b  ef  48  52  19  66  3c  16  e0  e3
 03                           |  ok | GET_SYSTEM_INFO
      55232 |     128960 | Tag |00  0f  ef  48  52  19  66  3c  16  e0  00
 00  1b  03  01  c9  c0       |  ok |
          0 |      22016 | Rdr |40  29  02  83  17
                              |  ok | WRITE_DSFID
      44480 |      44992 | Rdr |<EOF>
                             |     |
      98880 |     115264 | Tag |00  78  f0
                              |  ok |
[usb] pm3 -->

[image: K-001.png]

iceman1001, 2024-01-24:
in what way 
```
[usb] pm3 --> hf 15 writedsfid --dsfid 0x02

[+] Wrote DSFID 0x02 ( ok )
```

How are you verifying that dsfid isn't working?   Since the response from the card is saying ok.

w3a4gh69, 2024-01-25:
but after hf 15 info.
then nothing happens.

i have another build version
[image: K-002.png]

this version can change.

iceman1001, 2024-01-25:
Fixed with latest commit.

closed 2024-01-25 by iceman1001