RfidResearchGroup/proxmark3 #2275: Xerox tags can't be found via hf search, info command not working

JeroenSteen, 2024-01-25:
**Describe the bug**
Original and compatible Xerox HFD1 tags can't be found via commands: `hf search` and `hf xerox info`.

**Desktop:**
OS: Windows 11

hw version:
```
[ Proxmark3 RFID instrument ]

 [ Client ]
  Iceman/master/v4.17768-272-gf13d5c1b3-dirty-suspect 2024-01-25 15:51:13 822fd738b
  compiled with............. MinGW-w64 13.2.0
  platform.................. Windows (64b) / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... absent
  Python script support..... present
  Lua SWIG support.......... present
  Python SWIG support....... present

 [ Proxmark3 ]
  device.................... RDV4
  firmware.................. RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... absent

 [ ARM ]
  bootrom: Iceman/master/v4.17511-suspect 2023-11-13 10:19:09 db54c4c2c
       os: Iceman/master/v4.17768-272-gf13d5c1b3-dirty-suspect 2024-01-25 15:53:13 822fd738b
  compiled with GCC 12.2.0

 [ FPGA ]
  fpga_pm3_lf.ncd image 2s30vq100 2024-01-24 17:04:26
  fpga_pm3_hf.ncd image 2s30vq100 2024-01-24 17:04:37
  fpga_pm3_felica.ncd image 2s30vq100 2024-01-24 17:04:58
  fpga_pm3_hf_15.ncd image 2s30vq100 2024-01-24 17:04:48

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 69% used )
```
hw status:

```
[#] Memory
[#]   BigBuf_size............. 39796
[#]   Available memory........ 39796
[#] Tracing
[#]   tracing ................ 0
[#]   traceLen ............... 56
[#] Current FPGA image
[#]   mode.................... fpga_pm3_hf.ncd image 2s30vq100 2024-01-24 17:04:37
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... ok
[#]   Memory size............. 2 mbits / 256 kb
[#]   Unique ID (be).......... 0x39196F53175462D5
[#] Smart card module (ISO 7816)
[#]   version................. v3.10 ( Outdated )
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0
[#]
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A |
[#]     long leading reference |N/A | N/A | N/A | N/A | N/A | N/A | N/A |
[#]               leading zero |N/A | N/A | N/A | N/A | N/A | N/A | N/A |
[#]    1 of 4 coding reference |N/A | N/A | N/A | N/A | N/A | N/A | N/A |
[#]
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 264704
[#]   Transfer Speed PM3 -> Client... 529408 bytes/s
[#] Various
[#]   Max stack usage......... 5208 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... 8
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 31200 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[#] Flash memory dictionary loaded
[#]
```

data tune:

```
[=] ---------- Reminder ------------------------
[=] `hw tune` doesn't actively tune your antennas,
[=] it's only informative.
[=] Measuring antenna characteristics, please wait...
[-]  9
[=] ---------- LF Antenna ----------
[+] At 125.00 kHz .......... 75.01 V
[+] At 134.83 kHz .......... 34.30 V
[+] At 125.00 kHz optimal... 75.01 V
[+]
[+] Approx. Q factor measurement (*)
[+] Frequency bandwidth..... 12.6
[+] Peak voltage............ 13.1
[+] LF antenna ( ok )

[=] ---------- HF Antenna ----------
[+] 13.56 MHz............... 48.56 V
[+]
[+] Approx. Q factor measurement (*)
[+] peak voltage............ 8.5
[+] HF antenna ( ok )

(*) Q factor must be measured without tag on the antenna

[=] -------- LF tuning graph ---------
[+] Blue line  Divisor 88 / 134.83 kHz
[+] Red line   Divisor 95 / 125.00 kHz
```

**Additional context**
Xerox differs maybe from the 14B protocol, in the comments I can read: Xerox tag connect function: wup, anticoll, attrib, password (the original chips require all commands in this sequence), add framing enable flag. xerox chips use unframed commands during anticollision. I'm sorry but I can't find what's different in v4.17511 (where scanning Xerox tags works smoother).

iceman1001, 2024-01-25:
Run the following
```
-- OLD working firmware
hf xerox info
hf 14b list
trace save -f xerox_working

-- swap to LATEST firmware
hf xerox info
hf 14b list
trace save -f xerox_fail
```

post the output and share those two trace files.

JeroenSteen, 2024-01-26:
Here is the output of working and failing (with original used toner chip), with a zipped folder for the trace files. Traces of working and failing in one zipped file (since trace files are not allowed on Github): [xerox_traces.zip](https://github.com/RfidResearchGroup/proxmark3/files/14066476/xerox_traces.zip)

**Output of Xerox working (v4.17511):**

```
[=] downloading tracelog data from device
[+] Recorded activity (trace len = 1378 bytes)
[=] start = start of start frame end = end of frame. src = source of transfer
[=] ISO14443B - all times are in carrier periods (1/13.56MHz)

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |       2048 | Rdr |0d  37  21  92  f2                                                       |  ok | ?
      32544 |      34592 | Rdr |5d  37  21  71  71                                                       |  ok | ?
      35824 |      36080 | Rdr |b1                                                                       |     | ?
      37216 |      37472 | Rdr |b1                                                                       |     | ?
      38608 |      38864 | Rdr |b1                                                                       |     | ?
      39888 |      40144 | Rdr |a1                                                                       |     | ?
      41168 |      41424 | Rdr |a3                                                                       |     | ?
      42560 |      42816 | Rdr |b5                                                                       |     | ?
      43840 |      44096 | Rdr |a1                                                                       |     | ?
      45120 |      45376 | Rdr |a3                                                                       |     | ?
      46400 |      46656 | Rdr |a5                                                                       |     | ?
      47808 |      48064 | Rdr |b7                                                                       |     | ?
      49200 |      49456 | Rdr |b1                                                                       |     | ?
      50608 |      50864 | Rdr |b1                                                                       |     | ?
      52000 |      52256 | Rdr |b1                                                                       |     | ?
      53280 |      53536 | Rdr |a1                                                                       |     | ?
      54560 |      54816 | Rdr |a3                                                                       |     | ?
      55840 |      56096 | Rdr |a5                                                                       |     | ?
      57248 |      57504 | Rdr |b7                                                                       |     | ?
      58528 |      58784 | Rdr |a1                                                                       |     | ?
      59808 |      60064 | Rdr |a3                                                                       |     | ?
      61200 |      61456 | Rdr |b5                                                                       |     | ?
      62608 |      62864 | Rdr |b1                                                                       |     | ?
      64016 |      64272 | Rdr |b1                                                                       |     | ?
      65232 |      65488 | Rdr |a1                                                                       |     | ?
      66512 |      66768 | Rdr |b3                                                                       |     | ?
      67792 |      68048 | Rdr |b1                                                                       |     | ?
      69072 |      69328 | Rdr |b1                                                                       |     | ?
      70352 |      70608 | Rdr |b1                                                                       |     | ?
      71632 |      71888 | Rdr |b1                                                                       |     | ?
      72912 |      73168 | Rdr |b1                                                                       |     | ?
      74192 |      74448 | Rdr |b1                                                                       |     | ?
      75472 |      75728 | Rdr |b1                                                                       |     | ?
      76752 |      77008 | Rdr |b1                                                                       |     | ?
      78032 |      78288 | Rdr |b1                                                                       |     | ?
      79312 |      79568 | Rdr |b1                                                                       |     | ?
      80592 |      80848 | Rdr |b1                                                                       |     | ?
      81872 |      82128 | Rdr |b1                                                                       |     | ?
      83152 |      83408 | Rdr |b1                                                                       |     | ?
      84432 |      84688 | Rdr |b1                                                                       |     | ?
[... cut ...]
      32544 |      34592 | Rdr |5d  37  21  71  71                                                       |  ok | ?
          0 |       2048 | Rdr |0d  37  21  92  f2                                                       |  ok | ?
      32544 |      34592 | Rdr |5d  37  21  71  71                                                       |  ok | ?
          0 |       2048 | Rdr |0d  37  21  92  f2                                                       |  ok | ?
      32544 |      34592 | Rdr |5d  37  21  71  71                                                       |  ok | ?
```

iceman1001, 2024-01-26:
fixed

closed 2024-01-26 by iceman1001