RfidResearchGroup/proxmark3 #2278: USB device not recognized (Windows)

ioma8, 2024-01-27:
***Things to try before submitting bug report***
read the [troubleshooting guide](https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Troubleshooting.md)

***Compilation problems***
Try compiling with verbose.  `make VERBOSE=1` with main makefile or `make V=1` with cmake.

***flashing problems***
Have you followed the instructions properly?  ie,  flashed bootrom seperately first if you are going from Offical repo to Iceman repo.

-

**Describe the bug**
Hello, I compiled the master branch with 
PLATFORM=PM3GENERIC 
. I have Proxmark 3 easy 512. It stopped showing in my Windows Device Manager. Instead I have here entry "USB device not recognized." I tried both flashing via USB and via BUS pirate. Via USB I am no longer to even flash. I succeed flashing with Bus Pirate but the problem remains the same.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS:  Win 10
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
 - hw status
 - data tune

**Additional context**
Add any other context about the problem here.

ioma8, 2024-01-27:
Also I just tested this Proxmark on MacOs M1 and there it works flawlessly. It seems to be issue with the current master branch being usable (in the sense - device recognizable) in Windows 10. Seems somehow  related to drivers.

iceman1001, 2024-01-28:
How did you set  it up on windows?
Which guide did  you follow?
At which step does it fail?

ioma8, 2024-01-28:
I reflashed my Proxmark3 with current version of this repository. I did git clone of master branch. I copied Makefile.platform.sample into Makefile.platform and changed PLATFORM to PM3GENERIC. Compiled. Flashing worked okay (I tired flashing it both via USB and later also via JTAG interface with Bus Pirate).

With my older version my Proxmark 3 was recognized by Windows as COM serial device. After update my Windows sees it only as Unkonown USB device as per picture shown below. Thus I am unable to connect to it via pm3 command nor reflash it from Windows anymore (Windows no more sees it as COM device). But the device nor firmware is not corrupted. Under MacOs and Linux it shows up correctly under /dev/... and I am able to reflash it and also use pm3 commands succesfully. Thus it seems somehow related to Windows purely. Also I tried installing the Windows 7 driver onto this "Unknown USB device" but my Windows 10 wont allow it.

With earlier verions of rsg_proxmark (ie 2 years back when I last flashed it) it worked okay. After flashing it to current master it doesnt work.

[image]

iceman1001, 2024-01-28:
You will need to flash bootrom and then fullimage when going from older to latest. 

We also need you to fill in the detail of the issue template,   without proper details and output we are running blind and just guessing.

closed 2024-03-11 by iceman1001