RfidResearchGroup/proxmark3 #2279: `lf em 4x05 view` not working

hornet-hacker, 2024-01-27:
**Describe the bug**
When running the `lf em 4x05 view` command, it doesn't output correctly.  Also, when attempting to view a json file, it shows the full filepath, but when attempting to view a bin file it does not.  See output below

[usb] pm3 --> lf em 4x05 view -f A2.bin
[+] loaded 48 bytes from binary file `A2.bin`

[=] Addr | data     | ascii |lck| info
[=] -----+----------+-------+---+-----
[=] -----+----------+-------+---+-----

[=] Note:
[=] All ZEROS password block might be filler data
[usb] pm3 --> lf em 4x05 view -f A2.json
[+] loaded /media/user/XSYSTEM/files_prox/dump/A2.json

[=] Addr | data     | ascii |lck| info
[=] -----+----------+-------+---+-----
[=] -----+----------+-------+---+-----

[=] Note:
[=] All ZEROS password block might be filler data

**Expected behavior**
Here is the output of `cat` on the json in the above output:

┌──(user㉿dell11)-[/media/user/XSYSTEM/files_prox/dump]
└─$ cat A2.json
{
  "Created": "proxmark3",
  "FileType": "t55x7",
  "Card": {
    "ConfigBlock": "60107060"
  },
  "blocks": {
    "0": "60107060",
    "1": "1D555955",
    "2": "556A9955",
    "3": "99AAA6AA",
    "4": "00000000",
    "5": "00000000",
    "6": "00000000",
    "7": "00000000",
    "8": "60107060",
    "9": "E0150A24",
    "10": "6CDB0369",
    "11": "00000000"
  }
}                                                                                                                                                                               

**Desktop (please complete the following information):**
 - OS: kali linux
 - hw version
 [ Proxmark3 RFID instrument ]

 [ Client ]
  Iceman/master/v4.17768-310-g610408637-suspect 2024-01-27 12:11:41 9c8b8e61c
  compiled with............. GCC 13.2.0
  platform.................. Linux / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... present
  Python script support..... present
  Lua SWIG support.......... present
  Python SWIG support....... present

 [ Proxmark3 ]
  firmware.................. PM3 GENERIC
  external flash............ present
  FPC USART for BT add-on... present

 [ ARM ]
  bootrom: Iceman/master/v4.17768-204-g2d34713e5-suspect 2024-01-24 00:41:55 b91b1fb5d
       os: Iceman/master/v4.17768-310-g610408637-suspect 2024-01-27 12:11:56 9c8b8e61c
  compiled with GCC 12.2.1 20221205

 [ FPGA ] 
  fpga_pm3_lf.ncd image 2s30vq100 2024-01-24 17:04:26
  fpga_pm3_hf.ncd image 2s30vq100 2024-01-24 17:04:37
  fpga_pm3_felica.ncd image 2s30vq100 2024-01-24 17:04:58
  fpga_pm3_hf_15.ncd image 2s30vq100 2024-01-24 17:04:48

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 68% used )

 - hw status
[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 37556
[#]   Available memory........ 37556
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... fpga_pm3_hf.ncd image 2s30vq100 2024-01-24 17:04:37
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... ok
[#]   Unique ID (be).......... 0x383E18D3B7A468E1
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
[#]     long leading reference |  29 |  17 |  18 |  50 |  15 | N/A | N/A | 
[#]               leading zero |  29 |  17 |  18 |  40 |  15 | N/A | N/A | 
[#]    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 301568
[#]   Transfer Speed PM3 -> Client... 603136 bytes/s
[#] Various
[#]   Max stack usage......... 3520 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... -1
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 29364 Hz
[#] Installed StandAlone Mode
[#]   HF UNISNIFF, multimode HF sniffer with optional flashmem & runtime select (hazardousvoltage)
[#]   Compile-time default protocol: 14a
[#]   WITH_FLASH support.
[#] Flash memory dictionary loaded
[#]   Mifare.................. 1726 / 2047 keys
[#]   T55x7................... 123 / 1023 keys
[#]   iClass.................. 28 / 511 keys

**Additional context**
dumps made within the last couple of days, at the same time that the command that isn't working for me was being written.

iceman1001, 2024-01-28:
There was a fix pushed two days ago,  so try latest source.   Compile  / flash / test again.

regarding the file path printed or not , this is because we have not decided which way is better.
long paths take screen space,   but gives some info.
filename only is excellent for screen space,  but gives no info
See it as a quirk.

hornet-hacker, 2024-01-28:
I pulled and compiled latest source immediately before posting this issue.

iceman1001, 2024-01-28:
Well, I can't replicate your issue.    
Try running the commands with debug log enabled.

```data setd -2
lf em 4x05 dump
lf em 4x05 -f <file from above>

![image](https://github.com/RfidResearchGroup/proxmark3/assets/8577004/096b2cd3-fddd-4d25-930b-01472a1f5df4)

hornet-hacker, 2024-01-28:
I'm not sure how to turn on debugging, but the real issue I just discovered is named dump files.  If I create a dump and allow the client to name it, that dump loads fine with the view command.  If I then create a non-default named dump of the same card, the view command returns the broken output of my original comment.

hornet-hacker, 2024-01-28:
Further testing shows the view command will properly load a dump with the name `lf-4x05-*` regardless of the rest of the filename.
[image]

iceman1001, 2024-01-28:
Aha,
That is useful. 
We use filename to detect what kind of type of file it is.  
Let me have a look at it

iceman1001, 2024-01-28:
We don't have a simple way to detect what kind of tag it is. 
Hence we are falling back to filename to identify tag type.

iceman1001, 2024-01-28:
ok,   we did have better identification.

Pull latest and test,   should be fixed

hornet-hacker, 2024-01-28:
Fixed now.

closed 2024-01-28 by hornet-hacker