RfidResearchGroup/proxmark3 #2280: Python client not working

DjamikTea, 2024-01-28:
I followed the iceman's instructions in this video (https://www.youtube.com/watch?v=05v1A4pzZWQ) 

And when doing `./00make_swig.sh` in `client/experimental_lib/`

`Makefile:21: *** platform definitions have been changed, please "make clean" at the root of the project.  Stop.
Makefile:21: *** platform definitions have been changed, please "make clean" at the root of the project.  Stop.`

I did `make clean` and it didn't help.

Okay, I ignored it and went further according to the instructions.

Now when I reached `./02run_test.sh` I get the following error

`Traceback (most recent call last):
  File "/home/gopass/proxmark3/client/experimental_lib/example_py/./test.py", line 3, in <module>
    import pm3
  File "/home/gopass/proxmark3/client/pyscripts/pm3.py", line 15, in <module>
    import _pm3
ImportError: /home/gopass/proxmark3/client/experimental_lib/example_py/_pm3.so: undefined symbol: bf_generate`

**To Reproduce**
Steps to reproduce the behavior:
1) `sudo apt-get install --no-install-recommends git ca-certificates build-essential pkg-config libreadline-dev gcc-arm-none-eabi libnewlib-dev libbz2-dev liblz4-dev libpython3-dev libssl-dev`
2) `sudo apt install swig4.0`
3) `cd ~proxmark3/client/experimental_lib/`
4) `./00make_swig.sh`
5) `./01make_lib.sh`
6) `cd example_py/`
7) `rm -rf __pycache__/`
8) `./01link_lib.sh`
9) `./02run_test.sh`

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04.3 LTS (GNU/Linux 5.15.0-1045-raspi aarch64) (Raspberry Pi 4 4gb)
 - inside proxmark3 client run the following commands and paste the output here.
 
```
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ Client ]
  Iceman/master/v4.17768-310-g610408637-suspect 2024-01-27 15:58:47 9c8b8e61c
  compiled with............. GCC 11.4.0
  platform.................. Linux / aarch64
  Readline support.......... present
  QT GUI support............ absent
  native BT support......... absent
  Python script support..... present
  Lua SWIG support.......... present
  Python SWIG support....... present

 [ Proxmark3 ]
  device.................... device / fw mismatch
  firmware.................. RDV4
  external flash............ present
  smartcard reader.......... absent
  FPC USART for BT add-on... absent

 [ ARM ]
  bootrom: Iceman/master/v4.17768-281-g41d545f5b-suspect 2024-01-25 21:59:36 823c8c0ac
       os: Iceman/master/v4.17768-281-g41d545f5b-suspect 2024-01-25 22:02:10 823c8c0ac
  compiled with GCC 10.3.1 20210621 (release)

 [ FPGA ] 
  fpga_pm3_lf.ncd image 2s30vq100 2024-01-24 17:04:26
  fpga_pm3_hf.ncd image 2s30vq100 2024-01-24 17:04:37
  fpga_pm3_felica.ncd image 2s30vq100 2024-01-24 17:04:58
  fpga_pm3_hf_15.ncd image 2s30vq100 2024-01-24 17:04:48

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 69% used )

[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 39932
[#]   Available memory........ 39932
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... fpga_pm3_hf.ncd image 2s30vq100 2024-01-24 17:04:37
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... ok
[#]   Unique ID (be).......... 0x24A57F4B64B062DC
[#] Smart card module (ISO 7816)
[#]   version................. ( fail )
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
[#]     long leading reference |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
[#]               leading zero |  29 |  17 |  15 |  40 |  15 | N/A | N/A | 
[#]    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 358400
[#]   Transfer Speed PM3 -> Client... 716800 bytes/s
[#] Various
[#]   Max stack usage......... 3520 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... -1
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 32020 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[#] Flash memory dictionary loaded```

[usb] pm3 --> data tune
[=] ---------- Reminder ------------------------
[=] `hw tune` doesn't actively tune your antennas,
[=] it's only informative.
[=] Measuring antenna characteristics, please wait...
 🕛   9
[=] ---------- LF Antenna ----------
[!] ⚠️  LF antenna ( unusable )

[=] ---------- HF Antenna ----------
[+] 13.56 MHz............... 38.89 V
[+] 
[+] Approx. Q factor measurement (*)
[+] peak voltage............ 6.8
[+] HF antenna ( ok )

(*) Q factor must be measured without tag on the antenna

[-] ⛔ Not showing LF tuning graph since all values is zero.```

Thanks

iceman1001, 2024-01-28:
If you don't follow instructions its quite useless to help you. 

```
Makefile:21: *** platform definitions have been changed, please "make clean" at the root of the project.  Stop. 
Makefile:21: *** platform definitions have been changed, please "make clean" at the root of the project.  Stop.
```

closed 2024-01-28 by iceman1001

DjamikTea, 2024-01-28:
Sorry, I didn't really understand what you meant