RfidResearchGroup/proxmark3 #2286: undefined reference to `openpty'

tch1001, 2024-02-09:
I faced the following issue when running `make`. 

***Compilation problems***
```bash
$ make
undefined reference to `forkpty'
undefined reference to `openpty'
```

***Solution***
Append `-lutil` to the end of this line
```
proxmark3: $(OBJS) $(STATICLIBS) lualibs/pm3_cmd.lua lualibs/mfc_default_keys.lua
	$(info [=] CXX $@)
#	$(Q)$(CXX) $(PM3LDFLAGS) $(OBJS) $(STATICLIBS) $(LDLIBS) -o $@
	$(Q)$(CXX) $(PM3CFLAGS) $(PM3LDFLAGS) $(OBJS) $(STATICLIBS) $(LDLIBS) -o $@ -lutil
```

iceman1001, 2024-02-09:
This is a first....
Which OS are you running on?

tch1001, 2024-02-09:
Ubuntu 22.04

P.S. big fan of your work!

tch1001, 2024-02-09:
Yep! I am quite sure I did. It's alright then, you may ignore the pull request and I'll close this issue. Just wanted to open this issue so at least others who face the same bug can use my hacky fix :)

closed 2024-02-09 by tch1001

iceman1001, 2024-02-09:
I am confused since I never seen this message before. 

We don't even have any calls to pty.h,   or openpty or forkpty in the project.

You did `git clone`  and tried to compiled?     or have you downloaded this somewhere else?

tch1001, 2024-02-09:
Oh, the issue originated from `/usr/local/lib/libpython3.6m.a`, so I think it's a problem with my python installation. Sorry for the false reporting!