RfidResearchGroup/proxmark3 #2290: `hf emrtd dump` commands `--dir` argument handles user supplied directories as relative paths

Sanduuz, 2024-02-14:
**Describe the bug**
The `hf emrtd dump` commands `--dir` argument does not accept absolute paths, which leads to broken paths. Instead of relative paths it would be good that the user supplied dir would be handled as an absolute path. This way the user can also use relative paths themselves.

**To Reproduce**
Steps to reproduce the behavior:
1. Run the proxmark client software: `./pm3`
2. Run the following command: `hf emrtd dump -m [REDACTED] --dir /home/sanduuz/this/is/an/absolute/path`
3. Notice that saving files fail due to invalid path:
```
...
[=] ..
[=] Read EF_CardAccess , len 42
[!] ⚠️  file not found or locked `/home/sanduuz//home/sanduuz/this/is/an/absolute/path/EF_CardAccess.BIN`
[=] Authentication is enforced
[=] Switching to external authentication...
[=] External authentication with BAC successful.
...
```

**Expected behavior**
The path should be handled as an absolute path. This would mean that with the above command, the files should be saved under `/home/sanduuz/this/is/an/absolute/path/` instead of `/home/sanduuz//home/sanduuz/this/is/an/absolute/path/`.

**Version information:**
 - OS: Debian12 (`uname -a`: `Linux obsidian 6.1.0-17-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.69-1 (2023-12-30) x86_64 GNU/Linux`)
 - `hw version` and `hw status`
```
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ Client ]
  Iceman/master/v4.17768-389-gdac0a392b-suspect 2024-02-14 15:41:07 464278837
  compiled with............. GCC 12.2.0
  platform.................. Linux / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... present
  Python script support..... present
  Lua SWIG support.......... present
  Python SWIG support....... present

 [ Proxmark3 ]
  device.................... RDV4
  firmware.................. RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... absent

 [ ARM ]
  bootrom: Iceman/master/v4.17768-389-gdac0a392b-suspect 2024-02-14 15:41:21 464278837
       os: Iceman/master/v4.17768-389-gdac0a392b-suspect 2024-02-14 15:41:28 464278837
  compiled with GCC 12.2.1 20221205

 [ FPGA ] 
  fpga_pm3_lf.ncd image 2s30vq100 2024-02-03 15:12:10
  fpga_pm3_hf.ncd image 2s30vq100 2024-02-03 15:12:20
  fpga_pm3_felica.ncd image 2s30vq100 2024-02-03 15:12:41
  fpga_pm3_hf_15.ncd image 2s30vq100 2024-02-03 15:12:31

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 69% used )

[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size............. 39612
[#]   Available memory........ 39612
[#] Tracing
[#]   tracing ................ 0
[#]   traceLen ............... 35758
[#] Current FPGA image
[#]   mode.................... fpga_pm3_hf.ncd image 2s30vq100 2024-02-03 15:12:20
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... ok
[#]   Memory size............. 2 mbits / 256 kb
[#]   Unique ID (be).......... 0x238A0C97307C69D5
[#] Smart card module (ISO 7816)
[#]   version................. v3.10 ( Outdated )
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | n/a | n/a | 
[#]     long leading reference |n/a | n/a | n/a | n/a | n/a | n/a | n/a | 
[#]               leading zero |n/a | n/a | n/a | n/a | n/a | n/a | n/a | 
[#]    1 of 4 coding reference |n/a | n/a | n/a | n/a | n/a | n/a | n/a | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 177152
[#]   Transfer Speed PM3 -> Client... 354304 bytes/s
[#] Various
[#]   Max stack usage......... 3952 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... 30
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 31519 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[#] Flash memory dictionary loaded
[#] 
```

**Additional context**
I am not familiar with the codebase and styleguide (yet) so I won't butcher everything up by adding bad code. However, this might be a good point to start:
https://github.com/RfidResearchGroup/proxmark3/blob/dac0a392bd6f1caeb1893014d60b8096f7169ff1/client/src/cmdhfemrtd.c#L2192-L2195

iceman1001, 2024-02-14:
all file operations is done in `fileutils.c`    so you will see it should use those functions instead.   
it should handle paths,

iceman1001, 2026-09-12:
This bug is fixed.   Closing

closed 2026-09-12 by iceman1001