RfidResearchGroup/proxmark3 #2305: Error compiling on macOS related to Python

dotdeas, 2024-02-26:
**Describe the bug**
Issue with compiling latest commit

**To Reproduce**
Steps to reproduce the behavior:
1. Git clone
2. Compile (make clean && make -j8 all)
3. See error

`[=] CXX proxmark3
ld: Undefined symbols:
  _PyArg_UnpackTuple, referenced from:
      _SwigPyObject_own in pm3_pywrap.o
  _PyBool_FromLong, referenced from:
      _SwigPyObject_richcompare in pm3_pywrap.o
      _SwigPyObject_own in pm3_pywrap.o
  _PyBool_Type, referenced from:
      _SWIG_Python_UnpackTuple in pm3_pywrap.o
  _PyBytes_AsStringAndSize, referenced from:
      _SWIG_AsCharPtrAndSize in pm3_pywrap.o
  _PyCFunction_Type, referenced from:
      _SWIG_PyInstanceMethod_New in pm3_pywrap.o
      _SWIG_PyInstanceMethod_New in pm3_pywrap.o
      _pm3_swigregister in pm3_pywrap.o
      _pm3_swigregister in pm3_pywrap.o
      _SwigPyObject_dealloc in pm3_pywrap.o
      _SwigPyObject_dealloc in pm3_pywrap.o
      _SwigPyObject_dealloc in pm3_pywrap.o
      ...
  _PyCMethod_New, referenced from:
      _SWIG_PyInstanceMethod_New in pm3_pywrap.o
  _PyCapsule_GetPointer, referenced from:
      _SWIG_pchar_descriptor in pm3_pywrap.o
      _SWIG_Python_DestroyModule in pm3_pywrap.o
  _PyCapsule_Import, referenced from:
      _PyInit__pm3 in pm3_pywrap.o
      _SWIG_pchar_descriptor in pm3_pywrap.o
  _PyCapsule_New, referenced from:
      _PyInit__pm3 in pm3_pywrap.o
      _SWIG_pchar_descriptor in pm3_pywrap.o
  _PyConfig_Clear, referenced from:
      _CmdScriptRun in cmdscript.o
  _PyConfig_InitPythonConfig, referenced from:
      _CmdScriptRun in cmdscript.o
  _PyConfig_SetBytesArgv, referenced from:
      _CmdScriptRun in cmdscript.o
  _PyConfig_SetBytesString, referenced from:
      _CmdScriptRun in cmdscript.o
      _CmdScriptRun in cmdscript.o
  _PyDict_DelItemString, referenced from:
      _Pm3PyRun_SimpleFileNoExit in cmdscript.o
  _PyDict_GetItem, referenced from:
      _SWIG_pchar_descriptor in pm3_pywrap.o
  _PyDict_GetItemString, referenced from:
      _Pm3PyRun_SimpleFileNoExit in cmdscript.o
  _PyDict_New, referenced from:
      _PyInit__pm3 in pm3_pywrap.o
      _SWIG_Python_NewPointerObj in pm3_pywrap.o
      _SWIG_pchar_descriptor in pm3_pywrap.o
  _PyDict_SetItem, referenced from:
      _SWIG_pchar_descriptor in pm3_pywrap.o
  _PyDict_SetItemString, referenced from:
      _Pm3PyRun_SimpleFileNoExit in cmdscript.o
      _Pm3PyRun_SimpleFileNoExit in cmdscript.o
  _PyErr_Clear, referenced from:
      _Pm3PyRun_SimpleFileNoExit in cmdscript.o
      _Pm3PyRun_SimpleFileNoExit in cmdscript.o
      _PyInit__pm3 in pm3_pywrap.o
      _pm3_swigregister in pm3_pywrap.o
      _SWIG_pchar_descriptor in pm3_pywrap.o
[... 188 lines cut ...]
      _Pm3PyRun_SimpleFileNoExit in cmdscript.o
      _Pm3PyRun_SimpleFileNoExit in cmdscript.o
      _Pm3PyRun_SimpleFileNoExit in cmdscript.o
      ...
  __Py_NoneStruct, referenced from:
      _Pm3PyRun_SimpleFileNoExit in cmdscript.o
      __wrap_delete_pm3 in pm3_pywrap.o
      __wrap_delete_pm3 in pm3_pywrap.o
      __wrap_pm3_name_get in pm3_pywrap.o
      __wrap_pm3_name_get in pm3_pywrap.o
      _pm3_swigregister in pm3_pywrap.o
      _pm3_swiginit in pm3_pywrap.o
      ...
  __Py_NotImplementedStruct, referenced from:
      _SwigPyObject_richcompare in pm3_pywrap.o
      _SwigPyObject_richcompare in pm3_pywrap.o
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [proxmark3] Error 1
make: *** [client/all] Error 2`

**Desktop (please complete the following information):**
 - OS: macOS 14.3.1 (M1)

**Additional context**
Tried reinstalling Python 3.12.2, 3.12.1 and 3.12.0 but issue remains.

iceman1001, 2024-02-26:
There was the same issue raised on the pm3 homebrew GH repo where was found to be related to 3.12. 
The current solution is to force python3 v3.11 dependency. 

For OSx  feel free to find a solution since we don't use it.

skorokithakis, 2024-03-03:
I managed to compile with the following:

```
brew install python@3.11
```

```
diff --git a/client/Makefile b/client/Makefile
index 10929ffca..542a5cafa 100644
--- a/client/Makefile
+++ b/client/Makefile
@@ -269,8 +269,8 @@ endif
 
 ## Python3 (optional)
 ifneq ($(SKIPPYTHON),1)
-    PYTHONINCLUDES = $(shell $(PKG_CONFIG_ENV) pkg-config --cflags python3 2>/dev/null)
-    PYTHONLDLIBS = $(shell $(PKG_CONFIG_ENV) pkg-config --libs python3 2>/dev/null)
+    PYTHONINCLUDES = $(shell $(PKG_CONFIG_ENV) pkg-config --cflags python3.11 2>/dev/null)
+    PYTHONLDLIBS = $(shell $(PKG_CONFIG_ENV) pkg-config --libs python3.11 2>/dev/null)
     ifneq ($(PYTHONLDLIBS),)
         PYTHONLIBLD = $(PYTHONLDLIBS)
         PYTHONLIBINC = $(subst -I,-isystem ,$(PYTHONINCLUDES))
@@ -519,7 +519,7 @@ ifeq ($(SKIPPYTHON),1)
         $(info Python3 library:   skipped)
 else
     ifeq ($(PYTHON_FOUND),1)
-        $(info Python3 library:   Python3 v$(shell $(PKG_CONFIG_ENV) pkg-config --modversion python3) found, enabled)
+        $(info Python3 library:   Python3 v$(shell $(PKG_CONFIG_ENV) pkg-config --modversion python3.11) found, enabled)
     else
         $(info Python3 library:   Python3 not found, disabled)
     endif
```

It works (pm3 starts up, anyway, I didn't try to connect) after that.

AppleTechy, 2024-03-07:
I ended up just compiling by running `brew unlink python` which reverted back to python 3.9.x that ships with MacOS. I was then able to successfully run the compile (`brew install proxmark3` and then run `brew link python` to re- link to python 3.12.x

philicious, 2024-04-16:
works with `cmake ./client & make -j`

PR #2352 (2024-04-16): Fix building against Python3.12 on MacOS 
with make by prefering python3-embed over python3 (as CMakeLists.txt does see https://github.com/RfidResearchGroup/proxmark3/blob/master/client/CMakeLists.txt#L488-L500)

fixes #2305
closed 2024-04-16 by iceman1001