RfidResearchGroup/proxmark3 #2315: hf mfp chk -d flag always loads the default dictionary

snake-4, 2024-03-05:
https://github.com/RfidResearchGroup/proxmark3/blob/41696acc199fb8307590ccdad544b8bd7d7339d1/client/src/cmdhfmfp.c#L1507
https://github.com/RfidResearchGroup/proxmark3/blob/41696acc199fb8307590ccdad544b8bd7d7339d1/client/src/cmdhfmfp.c#L1513

`loadFileDICTIONARYEx` does not return PM3_SUCCESS if the buffer is too small so keyListLen is not set, which results in the default dictionary being loaded.

iceman1001, 2024-03-05:
Not sure what you are trying to do,  so below is the current output of running chk without a dictionary and with.

```
[usb] pm3 --> hf mfp chk
[=] Loaded 26 keys
[=] Search keys
.RRRRE

[=] -----+----------------------------------+----------------------------------
[=]  Sec | key A                            | key B
[=] -----+----------------------------------+----------------------------------
[=]  000 | -------------------------------- | --------------------------------
[=] -----+----------------------------------+----------------------------------

[usb] pm3 --> hf mfp chk -d client/dictionaries/mfp_default_keys.dic
[+] Loaded 32 keys from dictionary file `client/dictionaries/mfp_default_keys.dic`
[=] Loaded 26 keys
[=] Search keys
.RRRRE

[=] -----+----------------------------------+----------------------------------
[=]  Sec | key A                            | key B
[=] -----+----------------------------------+----------------------------------
[=]  000 | -------------------------------- | --------------------------------
[=] -----+----------------------------------+----------------------------------

[usb] pm3 -->
```

You can clearly see if called with a dictionary it reads it and uses it.

If you are trying to say that we always load the hard coded default key list ,  then the answer is yes.   There is a set of keys which is hard coded into the client.

snake-4, 2024-03-05:
The latter is what I meant, I thought the intended behavior was not to load the default dictionary as I thought the specified file would override it.

iceman1001, 2024-03-05:
We have this pattern across the pm3 client.  We also most always check against dedicated hard coded list of keys.
We not always inform the user of it.

closed 2024-03-05 by iceman1001