RfidResearchGroup/proxmark3 #2318: improved support for ISO14443-B tag info/dump

dhingos, 2024-03-09:
Testing an unknown RFID/NFC tag shows it is ISO14443-B as reported by pm3 command "hf search". pm3 recommends further use of the "hf 14b" commands. "hf 14b info" does not provide any tag information despite "hf search" reporting on the UID, ATQB, App Data and confirming the tag is compliant with the 14443-4 protocol. "hf 14b dump" responds with the message "Dumping Standard ISO-14443-B tags is not implemented yet". Additionally, I have to quit and restart pm3 in order to discover the the tag again using "hf search".

**Describe the solution you'd like**
Ability to dump an ISO 14443-B tag using the available "hf 14b dump" command. 

Thank you!

iceman1001, 2024-03-09:
We don't have many developers anymore,  so you are the best candidate to do it. 

Feel free to contribute to the project!

prodecom68, 2024-03-10:
The supposed solution does not work, this problem is not solved

prodecom68, 2024-03-10:
[image]

prodecom68, 2024-03-10:
What?
 I don't understand what robot means.

I just want to know what to do so that when I write the command, it works

dhingos, 2024-03-10:
> Why does this feel like a bot?

LOL It wasn't me, I swear!!! I do not know if I can contribute here, but I will give it a try....

burgueishon, 2024-10-27:
Help with this issue! I would like to write to a ISO14443-B

iceman1001, 2024-10-28:
Same answer as before.

closed 2025-06-08 by iceman1001