RfidResearchGroup/proxmark3 #2344: Linux compilation error [-Werror=format]

aoxhwjfoavdlhsvfpzha, 2024-04-11:
***Things to try before submitting bug report***
read the [troubleshooting guide](https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Troubleshooting.md)

***Compilation problems***
Try compiling with verbose.  `make VERBOSE=1` with main makefile or `make V=1` with cmake.

***flashing problems***
Have you followed the instructions properly?  ie,  flashed bootrom seperately first if you are going from Offical repo to Iceman repo.

-

**Describe the bug**
Compiling on Linux fails due to a type error right now, myself and two others at least have this issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Git pull the latest source
2. Make clean && make -j

**Expected behavior**
Successful make-ing

**Error Text:**
`src/proxguiqt.cpp:866:76:` error: format ‘%zu’ expects argument of type ‘size_t’, but argument 9 has type ‘uint32_t’ {aka ‘unsigned int’} [-Werror=format=]
  866 |     snprintf(str, sizeof(str), "@%u..%u  dt=%zi %szoom=%2.3f  CursorAPos=%zu  CursorBPos=%zu  GridX=%lf  GridY=%lf (%s) GridXoffset=%lf"`

**Desktop (please complete the following information):**
 - OS: Debian 12

**Additional context**
I could not replicate the problem within termux on a rooted phone, if that's of any help

bmt626, 2024-04-11:
I am running int to the same issue

jlitewski, 2024-04-11:
See PR #2343.

Iceman pushed changes to `proxguiqt.cpp` that also included some currently unreleased changes. As far as I know, he's working on getting it pushed.

Quick fix would be to change the `%zi` and `%zu` values it's complaining about to `%i` and `%u` locally and then compile.

iceman1001, 2024-04-11:
I am fully aware about the situation  and I was hoping to get out another refactoring but I didn't.

So I merged the PR to fixed it meanwhile.

closed 2024-04-11 by iceman1001