RfidResearchGroup/proxmark3 #2346: `--electra` does not write the correct additional data to a t55xx credential

Eltrick, 2024-04-12:
**Describe the bug**
When cloning an electra credential to a t55xx credential, incorrect data is written onto the t55 at blocks 3 and 4.

**To Reproduce**
1. Perform `lf em 410x clone --electra --id 04032dc7d9`
2. After `lf t55 det`, `lf t55 du --ns`
3. Inspect blocks 1 and 2 and see that it is the correct em 410 data
4. Inspect blocks 3 and 4, which are supposed to be the additional electra data wriiten to the t55, to see that what's actually written is the exact same as what's written on blocks 1 and 2, and not the additional data.

**Expected behavior**
Blocks 3 and 4 should've been written with the additional electra data (in this case, `7e1eaaaaaaaaaaaa`)

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: Kali Linux
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
 - hw status
 - data tune << (Please change this, I think the correct command should've been `hw tune`?)

[usb] pm3 --> hw ver

 [ Proxmark3 RFID instrument ]

 [ Client ]
  Iceman/master/v4.18341-45-g9a73e77d7-dirty-suspect 2024-04-12 04:57:05 201776ae5
  compiled with............. GCC 13.1.0
  platform.................. Linux / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... present
  Python script support..... present
  Lua SWIG support.......... present
  Python SWIG support....... present

 [ Proxmark3 ]
  firmware.................. PM3 GENERIC

 [ ARM ]
  bootrom: Iceman/master/v4.18341-45-g9a73e77d7-dirty-suspect 2024-04-12 04:56:31 201776ae5
       os: Iceman/master/v4.18341-45-g9a73e77d7-dirty-suspect 2024-04-12 04:57:54 201776ae5
  compiled with GCC 12.2.1 20221205

 [ FPGA ] 
  fpga_pm3_lf.ncd image 2s30vq100 2024-02-03 15:12:10
  fpga_pm3_hf.ncd image 2s30vq100 2024-02-03 15:12:20
  fpga_pm3_felica.ncd image 2s30vq100 2024-02-03 15:12:41
  fpga_pm3_hf_15.ncd image 2s30vq100 2024-02-03 15:12:31

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 62% used )

[usb] pm3 --> hw stat
[#] Memory
[#]   BigBuf_size............. 41500
[#]   Available memory........ 39192
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... fpga_pm3_lf.ncd image 2s30vq100 2024-02-03 15:12:10
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... no
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  31 |  20 |  18 |  50 |  15 | n/a | n/a | 
[#]     long leading reference |  31 |  20 |  18 |  50 |  15 | n/a | n/a | 
[#]               leading zero |  31 |  20 |  18 |  40 |  15 | n/a | n/a | 
[#]    1 of 4 coding reference |  31 |  20 |  18 |  34 |  15 |  50 |  66 | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 336384
[#]   Transfer Speed PM3 -> Client... 672768 bytes/s
[#] Various
[#]   Max stack usage......... 3880 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... -1
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 31503 Hz
[#] Installed StandAlone Mode
[#]   HF UNISNIFF - multimode HF sniffer (hazardousvoltage)
[#]   Compile-time default protocol... 14a
[#] 
[usb] pm3 --> hw tune

[=] -------- Reminder ----------------------------
[=] `hw tune` doesn't actively tune your antennas.
[=] It's only informative.
[=] Measuring antenna characteristics...
 🕛   9

[=] -------- LF Antenna ----------
[+] 125.00 kHz ........... 24.11 V
[+] 134.83 kHz ........... 16.23 V
[+] 120.00 kHz optimal.... 25.82 V
[+] 
[+] Approx. Q factor measurement
[+] Frequency bandwidth... 6.5
[+] Peak voltage.......... 7.5
[+] LF antenna............ ok

[=] -------- HF Antenna ----------
[+] 13.56 MHz............. 30.72 V
[+] 
[+] Approx. Q factor measurement
[+] Peak voltage.......... 8.9
[+] HF antenna ( ok )

[=] -------- LF tuning graph ------------
[+] Orange line - divisor 95 / 125.00 kHz
[+] Blue line - divisor   88 / 134.83 kHz

[=] Q factor must be measured without tag on the antenna

iceman1001, 2024-04-12:
And make sure you have a t5577 card which is full functional and not one of the ones which can only be written to block 0,1,2...

Yes,  those bad t5577 exists

Eltrick, 2024-04-13:
```
[usb] pm3 --> lf t55 det
[=]  Chip type......... T55x7
[=]  Modulation........ ASK
[=]  Bit rate.......... 2 - RF/32
[=]  Inverted.......... No
[=]  Offset............ 33
[=]  Seq. terminator... Yes
[=]  Block0............ 000880E0 (auto detect)
[=]  Downlink mode..... default/fixed bit length
[=]  Password set...... No

[usb] pm3 --> lf em 410x clone --electra --id 04032dc7d9
[+] Preparing to clone EM4102 to T55x7 tag with EM Tag ID 04032DC7D9 (RF/64)
[#] Clock rate: 64
[#] Tag T55x7 written with 0xff8120317787ee4e
[#] Electra 0x7e1eaaaaaaaaaaaa

[+] Done
[?] Hint: try `lf em 410x reader` to verify
[usb] pm3 --> lf t55 det
[=]  Chip type......... T55x7
[=]  Modulation........ ASK
[=]  Bit rate.......... 5 - RF/64
[=]  Inverted.......... No
[=]  Offset............ 33
[=]  Seq. terminator... Yes
[=]  Block0............ 00148080 (auto detect)
[=]  Downlink mode..... default/fixed bit length
[=]  Password set...... No

[usb] pm3 --> lf t55 du --ns

[=] ------------------------- T55xx tag memory -----------------------------

[+] Page 0
[+] blk | hex data | binary                           | ascii
[+] ----+----------+----------------------------------+-------
[+]  00 | 00148080 | 00000000000101001000000010000000 | ....
[+]  01 | FF812031 | 11111111100000010010000000110001 | .. 1
[+]  02 | 7787EE4E | 01110111100001111110111001001110 | w..N
[+]  03 | FF812031 | 11111111100000010010000000110001 | .. 1
[+]  04 | 7787EE4E | 01110111100001111110111001001110 | w..N
[+]  05 | 00000000 | 00000000000000000000000000000000 | ....
[+]  06 | 00000000 | 00000000000000000000000000000000 | ....
[+]  07 | 00000000 | 00000000000000000000000000000000 | ....

[+] Page 1
[+] blk | hex data | binary                           | ascii
[+] ----+----------+----------------------------------+-------
[+]  00 | 00148080 | 00000000000101001000000010000000 | ....
[+]  01 | E01500D0 | 11100000000101010000000011010000 | ....
[+]  02 | D3799D54 | 11010011011110011001110101010100 | .y.T
[+]  03 | 00A00003 | 00000000101000000000000000000011 | ....
[=] Called with no save option

[usb] pm3 --> lf t55 write -b 3 -d 7e1eaaaa
[=] Writing page 0  block: 03  data: 0x7E1EAAAA 
[usb] pm3 --> lf t55 write -b 4 -d aaaaaaaa
[=] Writing page 0  block: 04  data: 0xAAAAAAAA 
[usb] pm3 --> lf t55 du --ns

[=] ------------------------- T55xx tag memory -----------------------------

[+] Page 0
[+] blk | hex data | binary                           | ascii
[+] ----+----------+----------------------------------+-------
[+]  00 | 00148080 | 00000000000101001000000010000000 | ....
[+]  01 | FF812031 | 11111111100000010010000000110001 | .. 1
[+]  02 | 7787EE4E | 01110111100001111110111001001110 | w..N
[+]  03 | 7E1EAAAA | 01111110000111101010101010101010 | ~...
[+]  04 | AAAAAAAA | 10101010101010101010101010101010 | ....
[+]  05 | 00000000 | 00000000000000000000000000000000 | ....
[+]  06 | 00000000 | 00000000000000000000000000000000 | ....
[+]  07 | 00000000 | 00000000000000000000000000000000 | ....

[+] Page 1
[+] blk | hex data | binary                           | ascii
[+] ----+----------+----------------------------------+-------
[+]  00 | 00148080 | 00000000000101001000000010000000 | ....
[+]  01 | E01500D0 | 11100000000101010000000011010000 | ....
[+]  02 | D3799D54 | 11010011011110011001110101010100 | .y.T
[+]  03 | 00A00003 | 00000000101000000000000000000011 | ....
[=] Called with no save option
```

Eltrick, 2024-04-13:
As you can see, I am able to write further than just blocks 0, 1, 2.
If you also look at the t55's memory after the clone but before I manually write the electra data outputted, blocks 3&4 contain the exact same data as blocks 1&2, but they should have the data that I manually wrote, as shown in the t55's memory after* I manually wrote said data.

Eltrick, 2024-04-13:
As a sidenote, `lf search` doesn't seem to have electra detection yet, I can't find where the command is implemented so....

iceman1001, 2024-04-13:
Should work now and  feel free to add an electra descramble part in the em41x0 deocder

commit 6bc6ec23 (2024-04-13): fix #2346  refactored the write loop before it assumed a full t55xx memory layout.  Now handles block writes to individual parts
closed 2024-04-13 by iceman1001