RfidResearchGroup/proxmark3 #2350: Upgrade SWIG to 4.2.0 for Python 3.12 support

philicious, 2024-04-16:
**Describe the bug**
With 
> Python3 library:   Python3 v3.12 found, enabled
you cannot compile and get errors

```
ld: Undefined symbols:
  _PyArg_UnpackTuple, referenced from:
      _SwigPyObject_own in pm3_pywrap.o
  _PyBool_FromLong, referenced from:
...
alot more lines

The `pm3_pywrap.c` and `pm3_luawrap.c` files were generated with SWIG 4.0.2 as header comments suggests. SWIG 4.2.0 brings support for Python 3.12, according to https://www.swig.org/Release/RELEASENOTES

Thats also likely the reason for https://github.com/RfidResearchGroup/proxmark3/issues/2305

philicious, 2024-04-16:
interestingly its building against Python 3.12 using cmake instead of make. so its probably

iceman1001, 2024-04-16:
I updated the swig to 4.2.1 ...  
Now its down to testing if the experimental lib/client works :)

commit 6cc9c435 (2024-04-16): fix #2350  updated the swig wrappers to swig version 4.2.1.  Lets hope it solves the OSX issues
closed 2024-04-16 by iceman1001