RfidResearchGroup/proxmark3 #2372: Incompatible modules in standalone DANKARMULTI

francesco-scar, 2024-04-26:
**Describe the bug**
Some standalone modules are incompatible due to multiple definitions of the same functions, so we get compilation errors.

**To Reproduce**
Steps to reproduce the behavior:
1. Set `STANDALONE=DANKARMULTI`  in `Makefile.platform`
2. Edit `armsrc/Standalone/dankarmulti.c` similarly to the following:
```
#include "dankarmulti.h"
#define MODE_NAME em4100emul
#define MODE_FILE "lf_em4100emul.c"
#include "dankarmulti.h"
#define MODE_NAME em4100
#define MODE_FILE "lf_em4100rswb.c"
#include "dankarmulti.h"

START_MODE_LIST
ADD_MODE(em4100emul)
ADD_MODE(em4100)
END_MODE_LIST
```
3. Try to compile with make

**Expected behavior**
Compile successfully and get the two modules are available in standalone mode

**The problem**
I didn't searched exhaustively for the same problem with other modules, but with `lf_em4100emul` and `lf_em4100rswb` there are the same definitions of
```
static uint64_t rev_quads(uint64_t bits);
static void fill_buff(uint8_t bit);
static void construct_EM410x_emul(uint64_t id);
```

so just deleting them in the second file works fine.

Since dankarmulti actually includes the C source files in its own file, the static keyword doesn't help. Possible solutions are moving those common functions into a single, separate file (for example, `util`) or renaming each function differently in every module (maybe setting the standard to append the module name to each function or something similar). I don't know which one is the clearest solution.

ghiki, 2024-06-16:
Personally I have the same poble, for example with "icehid" and "tharexde", I solved easily  renaming functions and references 

in my case, in icehid.c, renamed all calls and fucntions itself from:
 append -> appendICEHID
DownloadLogInstructions ->DownloadLogInstructionsICEHID

iceman1001, 2024-06-16:
To be more consistent with naming convention,  try adding it as suffix instead.    icehid_xxxxx   vs  tharx__xxxx

iceman1001, 2025-02-20:
Closing since no interest in implementing the suggested fixes

closed 2025-02-20 by iceman1001