RfidResearchGroup/proxmark3 #2382: Build error: C++20 -Werror=template-id-cdtor in proxguiqt.cpp

AstreaeLuna, 2024-05-06:
**Describe the bug**
Build failure caused by C++20 compiler change: template-id not allowed for constructor in C++20

**To Reproduce**
Steps to reproduce the behavior:
1. Download git repo on Fedora 40
2. Install all required dependancies
3. Run `make -j all`
4. Build fails due to 
`/usr/include/qt5/QtCore/qfutureinterface.h:284:37: error: template-id not allowed for constructor in C++20 [-Werror=template-id-cdtor]
  284 |     explicit QFutureInterface<void>(State initialState = NoState)
      |                                     ^~~~~
/usr/include/qt5/QtCore/qfutureinterface.h:284:37: note: remove the ‘< >’`

**Expected behavior**
The build should succeed.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Linux - Fedora 40

**Additional context**
The error is resolved and the build is able to successfully complete if the suggest change is made and `<void>` is removed from Line 284 of /usr/include/qt5/QtCore/qfutureinterface.h

This leads me to believe that this isn't a problem strictly with Proxmark3 but with the QT development library. But I'm posting here to be sure.

Full build log is as follows:
```
===================================================================
Version info:      Iceman/master/v4.18341-152-g2bc7c5030
Platform name:     Proxmark3 generic target
PLATFORM:          PM3GENERIC
PLATFORM_FPGA:     xc2s30
PLATFORM_SIZE:     512
Platform extras:   No extra selected
Included options:  LF HITAG EM4x50 EM4x70 ZX8211 GENERAL_HF ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   LF_SAMYRUN
===================================================================
[*] MAKE client/all
[*] MAKE bootrom/all
[*] MAKE fpga_compress/all
[*] MAKE mfkey/all
[*] MAKE nonce2key/all
[*] MAKE mf_nonce_brute/all
[*] MAKE mfd_aes_brute/all
[*] MAKE cryptorf/all
[-] CC fpga_compress.c
[-] CC ../../common/lz4/lz4hc.c
[-] CC ../../common/lz4/lz4.c
[-] CC mf_nonce_brute.c
[-] CC nonce2key.c
[-] CC mfkey32.c
[-] CC ../../common/crapto1/crypto1.c
[-] CC ../../common/crapto1/crapto1.c
[-] CC ../../common/crapto1/crypto1.c
[-] CC ../../common/crapto1/crapto1.c
[-] CC ../../common/bucketsort.c
[-] CC ../../common/bucketsort.c
[-] CC brute_key.c
[-] CC cm.c
[-] CC ../../common/crapto1/crypto1.c
[-] CC ../../common/crapto1/crapto1.c
[-] CC nested_util.c
[-] CC ../../common/cryptorf/cryptolib.c
[-] CC util.c
[-] CC util_posix.c
[-] CC iso14443crc.c
[-] CC ../../common/bucketsort.c
[... 476 lines cut ...]
                 from src/proxgui.cpp:22:
/usr/include/qt5/QtCore/qfutureinterface.h:284:37: error: template-id not allowed for constructor in C++20 [-Werror=template-id-cdtor]
  284 |     explicit QFutureInterface<void>(State initialState = NoState)
      |                                     ^~~~~
/usr/include/qt5/QtCore/qfutureinterface.h:284:37: note: remove the ‘< >’
In file included from /usr/include/qt5/QtCore/qfuture.h:45,
                 from /usr/include/qt5/QtCore/QtCore:97,
                 from /usr/include/qt5/QtGui/QtGuiDepends:3,
                 from /usr/include/qt5/QtGui/QtGui:3,
                 from src/proxguiqt.h:30,
                 from src/proxguiqt.moc.cpp:10:
/usr/include/qt5/QtCore/qfutureinterface.h:284:37: error: template-id not allowed for constructor in C++20 [-Werror=template-id-cdtor]
  284 |     explicit QFutureInterface<void>(State initialState = NoState)
      |                                     ^~~~~
/usr/include/qt5/QtCore/qfutureinterface.h:284:37: note: remove the ‘< >’
[=] AR libmbedtls.a
cc1plus: all warnings being treated as errors
make[1]: *** [Makefile:1004: obj/proxguiqt.o] Error 1
make[1]: *** Waiting for unfinished jobs....
cc1plus: all warnings being treated as errors
make[1]: *** [Makefile:1004: obj/proxgui.o] Error 1
cc1plus: all warnings being treated as errors
make[1]: *** [Makefile:1004: obj/proxguiqt.moc.o] Error 1
make: *** [Makefile:177: client/all] Error 2
```

wh00hw, 2024-05-10:
same error on Arch, try to checkout the latest release tag then compile as usual.
`git checkout tags/v4.18341`

iceman1001, 2024-05-12:
This sounds more of a QT5 change than anything that we do

ikarus23, 2024-05-13:
Bumped into this issue as well.

On `v4.18341` the warning is:
```
In file included from /usr/include/qt/QtCore/qfuture.h:45,
                 from /usr/include/qt/QtCore/QtCore:97,
                 from /usr/include/qt/QtGui/QtGuiDepends:3,
                 from /usr/include/qt/QtGui/QtGui:3,
                 from src/proxguiqt.h:30,
                 from src/proxguiqt.cpp:19:
/usr/include/qt/QtCore/qfutureinterface.h:284:37: warning: template-id not allowed for constructor in C++20 [-Wtemplate-id-cdtor]
  284 |     explicit QFutureInterface<void>(State initialState = NoState)
      |                                     ^~~~~
/usr/include/qt/QtCore/qfutureinterface.h:284:37: note: remove the ‘< >’
```

On main the error is:
```
In file included from /usr/include/qt/QtCore/qfuture.h:45,
                 from /usr/include/qt/QtCore/QtCore:97,
                 from /usr/include/qt/QtGui/QtGuiDepends:3,
                 from /usr/include/qt/QtGui/QtGui:3,
                 from src/proxguiqt.h:30,
                 from src/proxguiqt.moc.cpp:10:
/usr/include/qt/QtCore/qfutureinterface.h:284:37: error: template-id not allowed for constructor in C++20 [-Werror=template-id-cdtor]
  284 |     explicit QFutureInterface<void>(State initialState = NoState)
      |                                     ^~~~~
/usr/include/qt/QtCore/qfutureinterface.h:284:37: note: remove the ‘< >’
```

So there is a difference in wording (error vs. warning). Not sure were it is coming from. Arch recently updated its gcc13 package. Maybe this changed something. Or like @iceman1001 said, something with Qt.

iceman1001, 2024-05-14:
funny thing,  the .moc.cpp is created by QT...    So its nothing we have control over more than it might be needed to regenerate on your systems?

AstreaeLuna, 2024-05-14:
@iceman1001 I'll look into reporting this bug to the QT devs instead. Out of curiosity, is there a way to retarget the build to use QT6?

iceman1001, 2024-05-14:
keep it open until you got some more answers.

sh7d, 2024-05-15:
For now you can use the following patch in order to just make build working and suppress the error:
```
diff --git a/client/Makefile b/client/Makefile
index 1b7090f68..22bd772b4 100644
--- a/client/Makefile
+++ b/client/Makefile
@@ -446,7 +446,7 @@ endif

 PM3CFLAGS += -DHAVE_SNPRINTF

-CXXFLAGS ?= -Wall -Werror
+CXXFLAGS ?= -Wall -Werror -Wno-error=template-id-cdtor
 CXXFLAGS += $(MYDEFS) $(MYCXXFLAGS) $(MYINCLUDES)

 PM3CXXFLAGS = $(CXXFLAGS)
```

AstreaeLuna, 2024-05-16:
I sent a bug report to the QT5 devs, who confirmed that the bug had been patched and the latest update to the upstream repositories should now resolve the issue.

https://bugzilla.redhat.com/show_bug.cgi?id=2280366

doegox, 2024-05-29:
If someone needs to patch their own Qt5 install, the line to change is https://src.fedoraproject.org/rpms/qt5-qtbase/blob/664f24855f26ccaee2501b26a50108b054bf19e2/f/qtbase-qfutureinterface-fix-build-with-gcc14.patch

```diff
 class QFutureInterface<void> : public QFutureInterfaceBase
 {
 public:
-    explicit QFutureInterface<void>(State initialState = NoState)
+    explicit QFutureInterface(State initialState = NoState)
         : QFutureInterfaceBase(initialState)
     { }
```
E.g. on Debian : `/usr/include/x86_64-linux-gnu/qt5/QtCore/qfutureinterface.h`

GladiatorX2016, 2024-07-23:
I simply replaced "`CONFIG += c++17`" with "`CONFIG += c++20`" in the project file.

iceman1001, 2024-07-24:
> I simply replaced "`CONFIG += c++17`" with "`CONFIG += c++20`" in the project file. And also added this: "`DEFINES += QT_DISABLE_DEPRECATED_BEFORE=0x060000`"

Interesting find,
You added the c++20 in the Client Makefile right?
and where did you add the QT define?

virtadpt, 2024-09-15:
You might also want to see this: https://aur.archlinux.org/packages/proxmark3-iceman-git

Some hand-hacking was involved insofar as compilation by an AUR helper was involved.

iceman1001, 2024-09-15:
Reminder, SKIPQT means you will not get the `data plot` window

You also mentioned: 
```
Deleting the substrings mfkey and 'nonce2key' from line 139 fixed that.
```
Where did you run into this?

virtadpt, 2024-09-16:
I know.  I sat and thought about it, and for what I'm using proxmark3 for right now that's fine.

I ran into that with the PKGBUILD file in the AUR.  It doesn't apply here, I think.

I had the same questions you did about the projectfile and QT definition, which is why I went off in a different direction (hopefully not a rabbithole).

iceman1001, 2024-09-16:
We have the following solutions: 

- @doegox   patch the QT file `qfutureinterface.h`
- @GladiatorX2016   patch the makefile  `CONFIG += c++20`  
- @sh7d  suppress the message   and the snprint issue which is something else

virtadpt, 2024-09-16:
I'll give them a try tonight, and if they work on my end I'll let the AUR package maintainer know.

doegox, 2024-11-26:
> If someone needs to patch their own Qt5 install, the line to change is https://src.fedoraproject.org/rpms/qt5-qtbase/blob/664f24855f26ccaee2501b26a50108b054bf19e2/f/qtbase-qfutureinterface-fix-build-with-gcc14.patch

FTR on Debian, the bug has been fixed in qtbase5-dev version 5.15.13+dfsg-4.
Current testing (Trixie) and unstable (Sid) are at 5.15.15.

iceman1001, 2025-02-20:
Is this still an issue?

Closing meanwhile.

ikarus23, 2025-02-20:
Can confirm, on Arch the error is no longer there. I guess like Debian, it got fixed.