RfidResearchGroup/proxmark3 #2383: hf mf autopwn: where do user keys and keys from hardcoded default array come from?

nico0481, 2024-05-09:
Using hf mf autopwn without supplied keys (-k) indicates 5 user keys loaded.
Where does it come from?
How to view those keys?

>[usb] pm3 --> hf mf autopwn
  [!] ⚠  no known key was supplied, key recovery might fail
  [+] loaded  5 user keys
  [+] loaded 61 keys from hardcoded default array

If i try to add a known key with '- k', user key count increments but i don't know what are the 5 user keys loaded.
If I use 'autopwn' all keys are found and final table shows found keys from dictionnary (D)
But if I try hf mf chk or fchk, with mfc_default_keys, it doesn't find unknown keys.

Found key with autopwn is not in mfc_default_keys dictionnary (checked with text editor)

unknown10777, 2024-05-09:
Got same result, didn't find where defined 5 user keys

nico0481, 2024-05-10:
After digging in the source code, correct me if I'm wrong:
I think signature of the card is recognized. Some keys on specific sectors are known (hardcoded in client source) this tell the 'signature' of the card, with this specific kind of card, it is possible to derive key from UID (with a specific algo). This computed derivative keys are maybe placed as "user keys" and used for dictionary attack.

iceman1001, 2024-05-12:
The default hard-coded list is a small very verified list of known default keys being used in the wild. 
The dictionary files is larger ones which might depend on your geographic / regional / system.  You can create a dictionary file based on which dump files you have.   There is a python script for that already.    

Does this make sense?

nico0481, 2024-05-12:
It's what I understood after reading the source code.
I was quite confused because it doesn't appear  when using hf mf chk or hf mf fchk.
Regards

iceman1001, 2024-05-13:
the hard coded default list is always used,  the custom user dictionary is optional.

closed 2024-05-13 by iceman1001