RfidResearchGroup/proxmark3 #2384: Use of free() in armsrc/desfire_crypto.c

jlitewski, 2024-05-12:
Now I'm not a smart man, but this could possibly cause problems, correct? Seems to have been here for about 4 years now.

https://github.com/RfidResearchGroup/proxmark3/blame/2bc7c5030234e43a1436d98bb7f5fec34802f29c/armsrc/desfire_crypto.c#L633

iceman1001, 2024-05-12:
Depends on what the data stored in that pointer does in that particular code path.  
If its not used,  free is ok to call onto it since its a local pointer.

henrygab, 2024-05-15:
@iceman1001 ... this does appear to be a memory corruption.

Here's one codepath:

* Variable `edata` is declared at [line 546](https://github.com/RfidResearchGroup/proxmark3/blame/2bc7c5030234e43a1436d98bb7f5fec34802f29c/armsrc/desfire_crypto.c#L546).
* `edata` stores allocation from `BigBuf_malloc()` at [line 580](https://github.com/RfidResearchGroup/proxmark3/blame/2bc7c5030234e43a1436d98bb7f5fec34802f29c/armsrc/desfire_crypto.c#L580)
* `break` at [line 598](https://github.com/RfidResearchGroup/proxmark3/blame/2bc7c5030234e43a1436d98bb7f5fec34802f29c/armsrc/desfire_crypto.c#L598) exits only the inner switch statement of [lines 566-631](https://github.com/RfidResearchGroup/proxmark3/blame/2bc7c5030234e43a1436d98bb7f5fec34802f29c/armsrc/desfire_crypto.c#L566-L631)
* `edata` is then passed to `free()`, even though it was not allocated by `alloc()`.

@jlitewski -- Looks to me like you found a memory corruption bug caused by mismatch allocator / deallocator.

**_Nice catch!_**

iceman1001, 2024-05-15:
Its even worse,   since we don't use any malloc/calloc,  but a bigbuffer alloc,     we never should use a call to `free`  
Its a bug.

commit 498af46f (2024-05-15): fix #2384 the use of free() is completely wrong as we use bigbuff allocations.  Converted the old style to new and removed the dynamic allocation.
closed 2024-05-15 by iceman1001