RfidResearchGroup/proxmark3 #2403: Simulated MIFARE Classic Card Responding Differently to 0x02 Command

sleepwalkera, 2024-06-13:
I’ve encountered an issue while using Proxmark3 to simulate a MIFARE Classic (M1) card on a specific reader. The simulated card does not work correctly with the reader, while the actual card works fine. Below, I have provided the logs for both the simulated card and the real card communication with the reader.

**Simulated Card Communication Log:**

```
      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |        352 | Rdr |02(2)                                                                    |     | 
       2612 |       8500 | Tag |02  00  00  AC  10                                                       |     | 
   17483194 |   17484250 | Rdr |26(7)                                                                    |     | REQA
   17485422 |   17487790 | Tag |04  00                                                                   |     | 
   20391518 |   20392574 | Rdr |26(7)                                                                    |     | REQA
   20393746 |   20396114 | Tag |04  00                                                                   |     | 
   28637332 |   28638388 | Rdr |26(7)                                                                    |     | REQA
   28639560 |   28641928 | Tag |04  00                                                                   |     | 
   36372700 |   36373756 | Rdr |26(7)                                                                    |     | REQA
   36374928 |   36377296 | Tag |04  00                                                                   |     | 
   39034626 |   39034978 | Rdr |02(2)                                                                    |     | 
   39037238 |   39043126 | Tag |02  00  00  AC  10                                                       |     | 
   55772118 |   55772470 | Rdr |02(2)                                                                    |     | 
   55774666 |   55780554 | Tag |02  00  00  AC  10                                                       |     | 
   56503810 |   56504866 | Rdr |26(7)                                                                    |     | REQA
   56506038 |   56508406 | Tag |04  00                                                                   |     | 
   86559708 |   86560060 | Rdr |02(2)                                                                    |     | 
   86562320 |   86568208 | Tag |02  00  00  AC  10                                                       |     | 
   89223774 |   89224830 | Rdr |26(7)                                                                    |     | REQA
   89226002 |   89228370 | Tag |04  00                                                                   |     | 
   97686394 |   97686746 | Rdr |02(2)                                                                    |     | 
   97688942 |   97694830 | Tag |02  00  00  AC  10                                                       |     | 
   98418518 |   98419574 | Rdr |26(7)                                                                    |     | REQA
   98420682 |   98423050 | Tag |04  00                                                                   |     | 
```

**Real Card Communication Log:**

```
      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |       2368 | Tag |04  00                                                                   |     | 
    2317532 |    2317884 | Rdr |02(2)                                                                    |     | 
    2319776 |    2322144 | Tag |04  00                                                                   |     | 
    2568956 |    2571420 | Rdr |93  20                                                                   |     | ANTICOLL
    2572592 |    2578480 | Tag |43  6E  CF  52  B0                                                       |     | 
    2833676 |    2844140 | Rdr |93  70  43  6E  CF  52  B0  45  27                                       |  ok | SELECT_UID
    2845376 |    2848896 | Tag |08  B6  DD                                                               |  ok | 
    3332156 |    3336860 | Rdr |60  27  48  2E                                                           |  ok | AUTH-A(39)
    3338480 |    3343216 | Tag |EE  FE  7F  01                                                           |     | AUTH: nt 
    3558748 |    3568124 | Rdr |73  B2  43  96! 33  79! 4D  21!                                          |     | AUTH: nr ar (enc)
    3569296 |    3574032 | Tag |a4! E8  71! 05!                                                          |     | AUTH: at (enc)
    3802364 |    3807132 | Rdr |25  AB  a9! A4                                                           |     | 
            |            |  *  |                                              key 013F0E139C0A prng WEAK |     |
            |            |  *  |30  24  24  CF                                                           |  ok | READBLOCK(36)
    3808304 |    3829168 | Tag |28! cf! a6! 23  61  58! E3  57  6F  6D  61! 06! EC  9D  9b! 97! 20  EF   |     | 
            |            |  *  |01  0B  B8  08  BB  B3  B0  03  4C  4C  FB  E9  FE  EB  E0  F0  3E  C9   |  ok | 
    5969996 |    5971052 | Rdr |26(7)                                                                    |     | REQA
    5972240 |    5974608 | Tag |04  00                                                                   |     | 
    6223324 |    6223548 | Rdr |01(1)                                                                    |     | 
    6226976 |    6232864 | Tag |43  6E  CF  52  B0                                                       |     | 
    6501680 |    6505200 | Tag |08  B6  DD                                                               |  ok | 
    8074032 |    8076400 | Tag |04  00                                                                   |     | 
    8325132 |    8327596 | Rdr |93  20                                                                   |     | ANTICOLL
    8328784 |    8334672 | Tag |43  6E  CF  52  B0                                                       |     | 
    8591260 |    8601724 | Rdr |93  70  43  6E  CF  52  B0  45  27                                       |  ok | SELECT_UID
    8602960 |    8606480 | Tag |08  B6  DD                                                               |  ok | 
   10416892 |   10417948 | Rdr |26(7)                                                                    |     | REQA
   10419136 |   10421504 | Tag |04  00                                                                   |     | 
   10670220 |   10670444 | Rdr |01(1)                                                                    |     | 
   10673872 |   10679760 | Tag |43  6E  CF  52  B0                                                       |     | 
   10936620 |   10947084 | Rdr |93  70  43  6E  CF  52  B0  45  27                                       |  ok | SELECT_UID
   10948320 |   10951840 | Tag |08  B6  DD                                                               |  ok | 
```

**Problem Summary:**

One key difference is the response to the `02` command. The real card responds with `04 00`, whereas the simulated card responds with `02 00 00 AC 10`. This discrepancy might be related to the simulated card failing on the reader, but I'm not entirely certain if this is the root cause.

**Details:**
- Firmware: Latest version of Proxmark3 (v4.18589)
- Hardware: Radiowar Black Edition

Could you please help me understand why this discrepancy occurs and how I can configure Proxmark3 to simulate the card correctly? Any guidance or suggestions for further troubleshooting steps would be greatly appreciated.

Thank you for your time and support!

sleepwalkera, 2024-07-05:
Recently, I used a Chameleon to emulate the same card, and it worked perfectly with this reader. It seems like a compatibility issue between the Proxmark3 emulated card and this specific reader. Any suggestions?

iceman1001, 2024-09-09:
Which commands did you use when running the simulation?

iceman1001, 2025-06-08:
Either your hardware or something else.   I can't replicate your issue. 
closing.

closed 2025-06-08 by iceman1001