RfidResearchGroup/proxmark3 #2405: [idea] Auto load "mf_user_keys.dic" from spiffs

ghiki, 2024-06-16:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

When you do some attack, to load the keys file (whe you know, for testing for example) is very repetitive to do "-d xxx.dic" or "mem load -f xxxx" before doing (obviously, this will do only when "ifdef WITH_FLASH")

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Dictionary file (or files) in spiffs that loads when starting any type of attack

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Load manually before with

    mem load -f mfc_default_keys -m         -> upload MFC keys
    mem load -f t55xx_default_pwds -t       -> upload T55XX passwords
    mem load -f iclass_default_keys -i      -> upload iCLASS keys

or passing to the attack command manually

**Additional context**
Add any other context or screenshots about the feature request here.

iceman1001, 2024-06-16:
you only need to upload the default dictionaries once on a RDV4 / spifffs based device.    Its stored in eeprom and read from eeprom when called in a function that takes advantage of that functionality.

ghiki, 2024-06-21:
Yep, but for example, a "hf mf autopwn" or any other, doesn't use any dic file although this file exists in the spiffs, the main idea is to use if the device has flash memory and a file with user Keys exist on It (now, if you want to use It, you need to load previously with "data" command)

iceman1001, 2024-06-22:
In that case are you thinking about updating `hf mf autopwn` to take advantage of the stored dictionary.

Uploading before to spiffs shouldn't affect autopwn to use it.  Like I said you only need to upload once. 
Well,  if the default dictionary file gets some additions you will of course re-upload it.   Doesn't happen that often.

jgoggan, 2024-07-12:
I really like this idea and thought the same thing the first time I started working with the PM3 hardware.  On the previous hardware that I used, it automatically saved any previously cracked information so that if you ran a crack again on the same card, it would immediately complete because it already had the key handy.  For the PM3, it always seemed funny to me that if I ran "hf mf autopwn" and it took a while and then cracked it -- and then I immediately ran "hf mf autopwn" again, it would do the entire crack again.  I would love it if every time it was successfully at cracking a key, it stored it to a local mf_user_keys.dic that was auto-loaded for any new crack attempt.  I mean, once it has the key, use it next time!  :)

iceman1001, 2024-07-12:
running attacks means running attacks,   if you store found keys which autopwn does,  you don't need to re-run attacks. 
You would use the already saved key file.

closed 2025-06-08 by iceman1001