RfidResearchGroup/proxmark3 #2410: Hardnested not successfull on PM3 rdv4 but on PM3 Easy (MFC EV1)

ikarus23, 2024-06-25:
Hi. I have (temporary) access to genuine MIFARE Classic EV1 tag. I tried to recover the keys using the hardnested method and noticed something strange: It works on the PM3 Easy but it does *NOT* work on the PM3 rdv4.

PM3 rdv4:
```
[usb] pm3 --> hf mf hardnested --blk 0 -a -k a0a1a2a3a4a5 --tblk 0 --tb
[=] Target block no   0, target key type: B, known target key: 000000000000 (not set)
[=] File action: none, Slow: No, Tests: 0
[=] Hardnested attack starting...
[=] ---------+---------+---------------------------------------------------------+-----------------+-------
[=]          |         |                                                         | Expected to brute force
[=]  Time    | #nonces | Activity                                                | #states         | time 
[=] ---------+---------+---------------------------------------------------------+-----------------+-------
[=]        0 |       0 | Start using 16 threads and AVX512F SIMD core            |                 |
[=]        0 |       0 | Brute force benchmark: 2065 million (2^30,9) keys/s     | 140737488355328 |   19h
[=]        0 |       0 | Loaded 0 RAW / 351 LZ4 / 0 BZ2 in 316 ms                | 140737488355328 |   19h
[=]        0 |       0 | Using 239 precalculated bitflip state tables            | 140737488355328 |   19h
[=]        3 |     112 | Apply bit flip properties                               |    102512648192 |   50s
[=]        4 |     224 | Apply bit flip properties                               |     30904090624 |   15s
[=]        5 |     336 | Apply bit flip properties                               |     22416209920 |   11s
[=]        6 |     448 | Apply bit flip properties                               |     21135179776 |   10s
[=]        7 |     560 | Apply bit flip properties                               |     21135179776 |   10s
[=]        8 |     671 | Apply bit flip properties                               |     21135179776 |   10s
[=]        9 |     779 | Apply bit flip properties                               |     21135179776 |   10s
[=]        9 |     890 | Apply bit flip properties                               |     21135179776 |   10s
[=]       10 |     997 | Apply bit flip properties                               |     21135179776 |   10s
[=]       11 |    1106 | Apply bit flip properties                               |     21135179776 |   10s
[=]       12 |    1216 | Apply bit flip properties                               |     21135179776 |   10s
[=]       13 |    1326 | Apply bit flip properties                               |     21135179776 |   10s
[=]       15 |    1437 | Apply Sum property. Sum(a0) = 128                       |      1825428480 |    1s
[=]       16 |    1548 | Apply bit flip properties                               |      1677347072 |    1s
[=]       17 |    1660 | Apply bit flip properties                               |      1677347072 |    1s
[=]       18 |    1770 | Apply bit flip properties                               |      1677347072 |    1s
[=]       18 |    1770 | (Ignoring Sum(a8) properties)                           |      1677347072 |    1s
```

PM3 Easy:
```
[usb] pm3 --> hf mf hardnested --blk 0 -a -k a0a1a2a3a4a5 --tblk 0 --tb
[=] Target block no   0, target key type: B, known target key: 000000000000 (not set)
[=] File action: none, Slow: No, Tests: 0
[=] Hardnested attack starting...
[=] ---------+---------+---------------------------------------------------------+-----------------+-------
[=]          |         |                                                         | Expected to brute force
[=]  Time    | #nonces | Activity                                                | #states         | time 
[=] ---------+---------+---------------------------------------------------------+-----------------+-------
[=]        0 |       0 | Start using 16 threads and AVX512F SIMD core            |                 |
[=]        0 |       0 | Brute force benchmark: 1933 million (2^30,8) keys/s     | 140737488355328 |   20h
[=]        0 |       0 | Loaded 0 RAW / 351 LZ4 / 0 BZ2 in 450 ms                | 140737488355328 |   20h
[=]        0 |       0 | Using 239 precalculated bitflip state tables            | 140737488355328 |   20h
[=]        3 |     112 | Apply bit flip properties                               |    248164073472 |  2min
[=]        5 |     224 | Apply bit flip properties                               |     24915030016 |   13s
[=]        6 |     335 | Apply bit flip properties                               |     21135179776 |   11s
[=]        6 |     447 | Apply bit flip properties                               |     21135179776 |   11s
[=]        7 |     558 | Apply bit flip properties                               |     21135179776 |   11s
[=]        8 |     669 | Apply bit flip properties                               |     21135179776 |   11s
[=]        9 |     780 | Apply bit flip properties                               |     21135179776 |   11s
[=]       10 |     891 | Apply bit flip properties                               |     21135179776 |   11s
[=]       11 |    1001 | Apply bit flip properties                               |     21135179776 |   11s
[=]       13 |    1110 | Apply Sum property. Sum(a0) = 128                       |      1887103872 |    1s
[=]       13 |    1222 | Apply bit flip properties                               |      1887103872 |    1s
[=]       14 |    1333 | Apply bit flip properties                               |      1887103872 |    1s
[=]       15 |    1441 | Apply bit flip properties                               |      1887103872 |    1s
[=]       16 |    1441 | (Ignoring Sum(a8) properties)                           |      1887103872 |    1s
[=]       18 |    1441 | Brute force phase completed.  Key found: XXXXXXXXXXXX   |               0 |    0s
```

I've tried a lot for the PM3 rdv4. Repeated over 10 times, changed the card position, changed the distance between card an PM3, etc., but it always pauses at ` (Ignoring Sum(a8) properties)` for some seconds and then fails.

Since it is so reproducible for this card, I think it might be a bug. I'm on arch using the latest code from the repo.

iceman1001, 2024-06-26:
Odd,  32 or 64b OS? 
I assume you compiled with clean and flashed properly.   Since I see AVX512F,   but there shouldn't be any difference in the hardnested loop client side based on device.   

Might be something with the device not answering a correct message for turning off,  which triggers this.  
Still it would be a long shot. 

Are you able to debug with GDB?

ikarus23, 2024-06-27:
OK, I was now able to crack the key using said arch system for the first time. The difference was, that the hardnested never hit `(Ignoring Sum(a8) properties)`. 

```
[=]       24 |    2421 | Apply bit flip properties                               |      1286970496 |    2s
[=]       25 |    2421 | (1. guess: Sum(a8) = 224)                               |      1286970496 |    2s
[=]       25 |    2421 | Apply Sum(a8) and all bytes bitflip properties          |      1286970496 |    2s
[=]       25 |    2421 | (2. guess: Sum(a8) = 128)                               |      1650445824 |    3s
[=]       26 |    2421 | Apply Sum(a8) and all bytes bitflip properties          |      1650445824 |    3s
[=]       27 |    2421 | (3. guess: Sum(a8) = 176)                               |      1366196096 |    2s
[=]       27 |    2421 | Apply Sum(a8) and all bytes bitflip properties          |      1366190464 |    2s
[=]       27 |    2421 | (4. guess: Sum(a8) = 144)                               |      1639314560 |    3s
[=]       27 |    2421 | Apply Sum(a8) and all bytes bitflip properties          |      1639302144 |    3s
[=]       27 |    2421 | (5. guess: Sum(a8) = 160)                               |      1915817728 |    3s
[=]       28 |    2421 | Apply Sum(a8) and all bytes bitflip properties          |      1915817728 |    3s
[=]       28 |    2421 | (6. guess: Sum(a8) = 192)                               |      2538548480 |    4s
[=]       28 |    2421 | Apply Sum(a8) and all bytes bitflip properties          |      2538546176 |    4s
[=]       28 |    2421 | (7. guess: Sum(a8) = 136)                               |      3514929152 |    6s
[=]       29 |    2421 | Apply Sum(a8) and all bytes bitflip properties          |      3514932480 |    6s
[=]       29 |    2421 | (8. guess: Sum(a8) = 200)                               |      5112785920 |    9s
[=]       29 |    2421 | Apply Sum(a8) and all bytes bitflip properties          |      5112780800 |    9s
[=]       29 |    2421 | (9. guess: Sum(a8) = 152)                               |      8961154048 |   16s
[=]       30 |    2421 | Apply Sum(a8) and all bytes bitflip properties          |      8961154048 |   16s
[=]       30 |    2421 | Brute force phase completed.  Key found: XXXXXXXX   |               0 |    0s
```

Also, as discussed over discord, It does not seem to be a PM3 rdv4 vs. PM3 Easy. The issue is probably with the system. I was able to crack that key using the same PM3 rdv4 on another arch device.

ikarus23, 2024-06-27:
OK, got some other news. I start to believe that the issue is with AVX512, like you expected. I have not Issues recovering the keys using the PM3 rdv4 on a system that has only AVX2. But I get the issue mentioned in the initial post on an system with AVX512.

Another quick test I did: I used a kali vm on the system with AVX512. Maybe the AVX512 feature is not exposed to the vm, because inside the vm everything is working fine with hardnested.

xianglin1998, 2024-09-20:
I also encountered the same problem, when using Inter's SDE tool to simulate AVX512 for HardNested decryption, I always couldn't hit the key correctly.

xianglin1998, 2024-09-20:
After some debugging, I found that the following lines of code seem to be missing support for avx512.

https://github.com/RfidResearchGroup/proxmark3/blob/c7af82f146180dc2655a529854dda07149b98501/client/deps/hardnested/hardnested_bf_core.c#L445

My change:
```c
#if MAX_BITSLICES > 64
                                && results.bytes64[1] == 0
#endif
#if MAX_BITSLICES > 128
                                && results.bytes64[2] == 0
                                && results.bytes64[3] == 0
#endif
#if MAX_BITSLICES > 256
                                && results.bytes64[4] == 0
                                && results.bytes64[5] == 0
                                && results.bytes64[6] == 0
                                && results.bytes64[7] == 0
#endif
```

By adding the above code, I can decrypt the correct key.

xianglin1998, 2024-09-20:
> OK, got some other news. I start to believe that the issue is with AVX512, like you expected. I have not Issues recovering the keys using the PM3 rdv4 on a system that has only AVX2. But I get the [...]

You can try to modify this file to add avx512 supported, If can work on your System with AVX512, I'll new a PR.

iceman1001, 2024-09-22:
The suggested fix works,   @nemanjan00   confirmed it yesterday.

closed 2024-09-22 by iceman1001

xianglin1998, 2024-09-22:
Perfect, Thanks for your work. 👍

ikarus23, 2024-09-22:
Nice catch! Thanks for fix. Now I can use it again on my AX512 work system.