RfidResearchGroup/proxmark3 #2418: Incorrect key displayed by mf_nonce_brute

Aptimex, 2024-07-11:
**Describe the bug**
I found an input for the `mf_nonce_brute` program that causes it to display the incorrect key even though it finds the correct key. Guessing it's a weird edge case in the code since I've only encountered this issue for this particular input. This input came from analyzing a trace of Mifare Classic 1k tag interacting with its reader. 

**To Reproduce**
1. Run `tools/mf_nonce_brute/mf_nonce_brute fcf77b54 1b456bdd 1110 f215b6 f9eb95e9 0011 bf55d0b1 0000 AAD4126B`
2. Notice that it identifies `Key candidate [ a70d37afcc2b ]`
3. But the final line says `Valid Key found [ 7c2337afcc2b ]`

I verified that `a70d37afcc2b` is the correct key in this case (it correctly decrypts the associated block). The final line is just displaying an incorrect value for some reason. 

**Expected behavior**
The final line of the output should say `Valid Key found [ a70d37afcc2b ]` (same as the `Key candidate` found previously in the output)

**Screenshots**
Full command and output I get:
```
tools/mf_nonce_brute/mf_nonce_brute fcf77b54 1b456bdd 1110 f215b6 f9eb95e9 0011 bf55d0b1 0000 AAD4126B

Mifare classic nested auth key recovery

----------- information ------------------------
uid.................. fcf77b54
nt encrypted......... 1b456bdd
nt parity err........ 1110
nr encrypted......... 00f215b6
ar encrypted......... f9eb95e9
ar parity err........ 0011
at encrypted......... bf55d0b1
at parity err........ 0000
next encrypted cmd... AAD4126B

Bruteforce using 12 threads

----------- Phase 1 pre-processing ------------------------
Testing default keys using NESTED authentication...

----------- Phase 2 examine -------------------------------
Looking for the last bytes of the encrypted tagnonce

Target old MFC...
CMD enc( aad4126b )
    dec( 302424cf )    <-- valid cmd

Key candidate [ ....37afcc2b ]
Key candidate [ a70d37afcc2b ]

execution time 0.40 sec

----------- Phase 3 validating ----------------------------
uid.................. fcf77b54
partial key.......... 37afcc2b
nt enc............... 1b456bdd
nr enc............... 00f215b6
next encrypted cmd... AAD4126B

Looking for the upper 16 bits of the key

enc:  AAD4126B
dec:  610BFEDC

Valid Key found [ 7c2337afcc2b ]
```

**Desktop (please complete the following information):**
 - OS: Windows (running in WSL v1, Ubuntu 22.04.4 LTS)
 - hw version
 ```
 pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ Client ]
  Iceman/master/v4.18589-69-gef97dd3f7-suspect 2024-07-08 22:26:32 f1dc6862f
  compiled with............. GCC 11.4.0
  platform.................. Linux / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... present
  Python script support..... present
  Lua SWIG support.......... present
  Python SWIG support....... present

 [ Proxmark3 ]
```
 - hw status
 ```
hw status
[#] Memory
[#]   BigBuf_size............. 37620
[#]   Available memory........ 37620
[#] Tracing
[#]   tracing ................ 1
[#]   traceLen ............... 0
[#] Current FPGA image
[#]   mode.................... fpga_pm3_hf.ncd image 2s30vq100 2024-02-03 15:12:20
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... ok
[#]   Memory size............. 2 mbits / 256 kb
[#]   Unique ID (be).......... 0x2A5B26DF230C69D5
[#] Smart card module (ISO 7816)
[#]   version................. v3.11 ( Outdated )
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0
[#]
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | n/a | n/a |
[#]     long leading reference |  29 |  17 |  15 |  47 |  15 | n/a | n/a |
[#]               leading zero |  29 |  17 |  15 |  40 |  15 | n/a | n/a |
[#]    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 |
[#]
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 503ms
[#]   Bytes transferred.............. 5632
[#]   Transfer Speed PM3 -> Client... 11196 bytes/s
[#] Various
[#]   Max stack usage......... 3520 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... -1
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 31896 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[#] Flash memory dictionary loaded
[#]   T55x7................... 109 / 1023 keys
[#]   iClass.................. 7 / 511 keys
```
 - data tune (assuming you meant `hw tune` since `data tune` is not a valid command)
```
hw tune

[=] -------- Reminder ----------------------------
[=] `hw tune` doesn't actively tune your antennas.
[=] It's only informative.
[=] Measuring antenna characteristics...
 🕗  13

[=] -------- LF Antenna ----------
[+] 125.00 kHz ........... 71.36 V
[+] 134.83 kHz ........... 40.11 V
[+] 126.32 kHz optimal.... 75.91 V
[+]
[+] Approx. Q factor measurement
[+] Frequency bandwidth... 12.7
[+] Peak voltage.......... 13.2
[+] LF antenna............ ok

[=] -------- HF Antenna ----------
[+] 13.56 MHz............. 48.65 V
[+]
[+] Approx. Q factor measurement
[+] Peak voltage.......... 8.5
[+] HF antenna ( ok )

[=] -------- LF tuning graph ------------
[+] Orange line - divisor 95 / 125.00 kHz
[+] Blue line - divisor   88 / 134.83 kHz

[=] Q factor must be measured without tag on the antenna
```

**Additional context**
If you need the original trace file let me know and I can include it.

iceman1001, 2024-07-11:
Thanks,

the reason for the final candidate is mentioned right above it.   
When bruteforcing the 16 bits left we try to find a valid combo of 4 byte MFC command.

In this case the decrypted response is  61 0B ,,  meaning auth - key b, to block 0x0b.   and it has a valid CRC 14a of 0xFEDC
Meaning all four bytes was decrypted into something meaningful and valid in MFC protocol. 

The question is of course if it printed the wrong key...

```

Looking for the upper 16 bits of the key

enc:  AAD4126B
dec:  610BFEDC

Valid Key found [ 7c2337afcc2b ]
```

iceman1001, 2024-07-11:
and you should update your sim module firmware...
```
[#] Smart card module (ISO 7816)
[#]   version................. v3.11 ( Outdated )
```

iceman1001, 2024-07-11:
Can you give me a output of all keys from in the memory dump of that card?

iceman1001, 2024-07-11:
Pull latest and see the updated output.

Its an odd case indeed.

closed 2024-07-11 by iceman1001

Aptimex, 2024-07-12:
Thanks for looking into this, new output does a good job clarifying what's going on in this situation. 

I see you closed this, but if you still want the full key dump from the card let me know.

iceman1001, 2024-07-12:
Nay I am fine, 
your sample to run the command was enough for me to figure out what was going on and why.

commit 05b50a6c (2024-07-11): fix #2418 - the tool mf_nonce_brute has in some odd cases when bruteforcing the last upper 16 bits a chance of actually decrypt the four bytes into a valid mifare classic protocol command with a valid ISO14443-a CRC.  See github issue for example.\n  We now bruteforce all 0xFFFF keyspace and keep track of how many candidates was found.  The output has been improved to help user in this case too.