RfidResearchGroup/proxmark3 #2437: LF SEARCH SEGMENTATION FAULT ERROR

radiosamurai, 2024-07-31:
Hello i am trying to read some button type rfid system i think it is LF because when i do LF SEARCH thing i got

 [=] Note: False Positives ARE possible
[=]
[=] Checking for known tags...
[=]
pm3: line 249:  1279 Segmentation fault      $CLIENT "$@"

error when i try another LF card it will read with success i dont know what type of RFID card is.

radiosamurai, 2024-07-31:
+ i also tried another OS like macos and linux too it gives this error when i read that LF tag.

iceman1001, 2024-08-01:
You would need to fill in more information about your situation,

What is your proxmark3 model?
Are you running latest source compiled and flashed for your proxmark3 model?
What is the output of the `hw tune
Which is the `LF card` you are trying to read?`

radiosamurai, 2024-08-01:
> You would need to fill in more information about your situation,
> 
> What is your proxmark3 model? Are you running latest source compiled and flashed for your proxmark3 model? What is the output of the `hw tune Which is the `LF card` you are trying to read?`

i buy from aliexpress it maybe proxmark3 easy rev 3.0? yes i flashed iceman firmware inside of it i think the LF card is em4x50 because i tried every protocol the iceman supports. i can read when i do em4x50 reader but when i try lf search on this thing it just gives segmentation fault error. i am not compiled hole source i just installed from release.

iceman1001, 2024-08-02:
Please answer the questions.

iceman1001, 2024-09-09:
There were some changes,  do you still experience crashes running `lf search` ?

radiosamurai, 2024-09-09:
i will try that.

tcfsoft, 2024-09-19:
I have the same issue with latest versions, here's the hw tune info of my proxmark:

```
[usb] pm3 --> hw tune

[=] -------- Reminder ----------------------------
[=] `hw tune` doesn't actively tune your antennas.
[=] It's only informative.
[=] Measuring antenna characteristics...
[/] 10

[=] -------- LF Antenna ----------
[+] 125.00 kHz ........... 25.64 V
[+] 134.83 kHz ........... 17.61 V
[+] 122.45 kHz optimal.... 26.18 V
[+]
[+] Approx. Q factor measurement
[+] Frequency bandwidth... 6.7
[+] Peak voltage.......... 7.6
[+] LF antenna............ ok

[=] -------- HF Antenna ----------
[+] 13.56 MHz............. 14.82 V
[+]
[+] Approx. Q factor measurement
[+] Peak voltage.......... 4.3
[+] HF antenna ( ok )

[=] -------- LF tuning graph ------------
[+] Orange line - divisor 95 / 125.00 kHz
[+] Blue line - divisor   88 / 134.83 kHz

[=] Q factor must be measured without tag on the antenna
```
I can "lf em 4x50 info/dump/write" with no problems.

Only reading the block or doing lf search gives the error Segmentation fault:

```
[usb] pm3 --> lf search

[=] Note: False Positives ARE possible
[=]
[=] Checking for known tags...
[=]
pm3: line 249:   445 Segmentation fault      $CLIENT "$@"
Press any key to continue . . .
```

```
[usb] pm3 --> lf em 4x50 rdbl -b 5
pm3: line 249:  1986 Segmentation fault      $CLIENT "$@"
Press any key to continue . . .

```

I tried with an older version I was using before upgrading (Iceman/master/v4.18341-45-g9a73e77d7) and works with no error ok if you want to check 👍

```
[usb] pm3 --> lf search

[=] Note: False Positives ARE possible
[=]
[=] Checking for known tags...
[=]

[=]   # | word (msb)  | word (lsb)  | desc
[=] ----+-------------+-------------+--------------------
[=]  32 | 6D 35 16 C0 | 03 68 ac b6 | device serial number ( RO )
[=] ----+-------------+-------------+--------------------
[=]  Serial: 6D 35 16 C0

[+] Valid EM4x50 ID found!

[usb] pm3 -->

```

radiosamurai, 2024-09-19:
yes problem seems to fixed

closed 2024-09-19 by radiosamurai