RfidResearchGroup/proxmark3 #2440: make install fails because mfc tools could not be found

alex-kozin, 2024-08-03:
MFC tools were moved under tools/mfc in [this commit](https://github.com/RfidResearchGroup/proxmark3/commit/c47578c04884cfa5ae5ec9830492d6e37cb3bf5f) but the relative path was not updated in Makefile when running install target:

https://github.com/RfidResearchGroup/proxmark3/blob/ee8b9ca74b10536db246286179183e0f1d89770e/Makefile#L36

Resulting in the following failure:

```
*] MAKE mfc_card_reader/install
[@] Installing mfkey32 mfkey32v2 mfkey64 mf_nonce_brute mf_trace_brute to /usr/local...
[*] MAKE mfd_aes_brute/install
[@] Installing brute_key mfd_aes_brute mfd_multi_brute to /usr/local...
[*] MAKE cryptorf/install
[@] Installing cm sm sma sma_multi to /usr/local...
[@] Installing common resources to /usr/local...
cp: cannot stat 'tools/pm3_eml2lower.sh': No such file or directory
cp: cannot stat 'tools/pm3_eml2upper.sh': No such file or directory
cp: cannot stat 'tools/pm3_mfdread.py': No such file or directory
cp: cannot stat 'tools/pm3_mfd2eml.py': No such file or directory
cp: cannot stat 'tools/pm3_eml2mfd.py': No such file or directory
make: *** [Makefile:53: common/install] Error 1
```

Happy to open a PR myself, would be my first contribution, but I will let maintainers chime in on this first in case I am overstepping here. Thanks.

clipboard1, 2024-08-06:
I'm glad to see your problem, I was wondering what the problem was... But now I know that I'm not alone huh

iceman1001, 2024-08-10:
@doegox   Something you want to clean up?

doegox, 2024-08-10:
Ha yes I saw it myself and fixed it yesterday

closed 2024-08-10 by doegox