RfidResearchGroup/proxmark3 #2451: [idea]About fm11rf08s

hanyuhang, 2024-08-16:
Now,we can use the scripts of 'fm11rf08s_recovery.py' in the Proxspace , but if I compile the projects, the script don't work.

iceman1001, 2024-08-16:
You will need to write more about what goes wrong.  Give sample of output.

hanyuhang, 2024-08-16:
The log：
[usb] pm3 --> script run fm11rf08s_recovery
[+] executing python D:\proxmark3-20240815-74db7a22b82e33f48ebf57239f6526d41954c1a0\client\pyscripts/fm11rf08s_recovery.py
[+] args ''
Could not find platform independent libraries <prefix>
Could not find platform dependent libraries <exec_prefix>
Python path configuration:
  PYTHONHOME = (not set)
  PYTHONPATH = (not set)
  program name = 'D:/proxmark3-20240815-74db7a22b82e33f48ebf57239f6526d41954c1a0/client/pyscripts/fm11rf08s_recovery.py'
  isolated = 0
  environment = 0
  user site = 1
  safe_path = 0
  import site = 1
  is in build tree = 0
  stdlib dir = 'D:/a/msys64/mingw64/lib/python3.11'
  sys._base_executable = 'D:/proxmark3-20240815-74db7a22b82e33f48ebf57239f6526d41954c1a0/client/pyscripts/fm11rf08s_recovery.py'
  sys.base_prefix = 'D:/a/msys64/mingw64'
  sys.base_exec_prefix = 'D:/a/msys64/mingw64'
  sys.platlibdir = 'lib'
  sys.executable = 'D:/proxmark3-20240815-74db7a22b82e33f48ebf57239f6526d41954c1a0/client/pyscripts/fm11rf08s_recovery.py'
  sys.prefix = 'D:/a/msys64/mingw64'
  sys.exec_prefix = 'D:/a/msys64/mingw64'
  sys.path = [
    'D:/a/msys64/mingw64/lib/python311.zip',
    'D:/a/msys64/mingw64/lib/python3.11',
    'D:/a/msys64/mingw64/lib/python3.11/lib-dynload',
[image]

iceman1001, 2024-08-16:
Sounds like a proxspace / mingw python issue  and not related to the actual code.

closed 2024-08-16 by iceman1001

tec1987, 2024-11-24:
Do't use "autobuild.bat".
Run runme64.bat
use pacman -S <name or part of the name of the package>  install depend packge
pacman -Syuu   to update mysys2

Then refer installation instructions to continue
https://github.com/Gator96100/ProxSpace?tab=readme-ov-file#installation