RfidResearchGroup/proxmark3 #2457: FM11NT021 Compatible NTAG213 tag is not detected

tywtyw2002, 2024-08-20:
FUDAN FM11NT021 its compatibility with the NXP Ntag 213.

[FM11NT021_sds_eng.pdf](https://github.com/user-attachments/files/16667599/FM11NT021_sds_eng.pdf)

The Proxmark3 detected this tag as NTAG203 due to the FM11NT021 does not have `0x60` `GET_VERSION` Command.

I did some work around, by check the page 44 in `GetHF14AMfU_Type` function. (203 does not has page 44, but 213 has.)

```
                // need to re-select after authentication error
                if (ul_select(&card) == false) {
                    return MFU_TT_UL_ERROR;
                }

                uint8_t data[16] = {0x00};
                // read page 0x26-0x29 (last valid ntag203 page)
                status = ul_read(0x26, data, sizeof(data));
                if (status <= 1) {
                    tagtype = MFU_TT_UL;
                } else {
                    // read page 0x30 (should error if it is a ntag203)
                    status = ul_read(0x30, data, sizeof(data));
                    if (status <= 1) {
						// check page 44 for fudan ntag213.
                        status = ul_read(0x2C, data, sizeof(data));
                        if (status <= 1) {
                            tagtype = MFU_TT_NTAG_213;
                        }else {
                            tagtype = MFU_TT_NTAG_203;
                        }
                    } else {
                        tagtype = MFU_TT_UNKNOWN;
                    }
                }
```

iceman1001, 2024-08-20:
Cool!

There is so many tags,  not sure it should be identified as 213 but as a 213 compatible.   Which would mean more changes. 

Meanwhile make a PR !

iceman1001, 2024-09-10:
Pull latest and see if its detected properly

closed 2024-09-10 by iceman1001