RfidResearchGroup/proxmark3 #2494: [RELEASE 4.18981] Checklist - release name "Backdoor"

iceman1001, 2024-09-09:
# Checklist

- [ ] `make style`
- [ ] `make miscchecks`
- [ ] `make clean; make client CC=clang CXX=clang++ LD=clang++` on recent Debian or Ubuntu
- [ ] `mymanualchecks.sh`
- [ ] `mycppcheck.sh` no alarming warning?
- [ ] `tools/build_all_firmwares.sh` check that the script contains all standalone modes then compile all standalone modes (linux only)
- [ ] `experimental_lib` compilation & tests
- [ ] `experimental_client_with_swig` compilation & tests
- [ ] GitHub Actions - green across the board ( MacOS, Ubuntu, Windows)

# OS compilation and tests

Run `tools/release_tests.sh` on:

- [ ] RPI Zero
- [ ] Jetson Nano
- [ ] WSL
- [ ] PSv3.xx
- [ ] Archlinux
- [ ] Kali
- [ ] Debian Stable
- [ ] Debian Testing
- [ ] Ubuntu 22
- [ ] ParrotOS
- [ ] Fedora 37
- [ ] OpenSuse Leap
- [ ] OpenSuse Tumbleweed
- [ ] OSX (MacPorts)
- [ ] OSX (Homebrew)
- [ ] Android
- [ ] Termux

# creating release

- [ ] CHANGELOG.md: add title: `## [Backdoor][YYYY-MM-DD]`
- [ ] `make release RELEASE_NAME="Backdoor"`
  - last line of output gives you next command to run.
  - Sample:  `git push && git push origin v4.18981`
- [ ] CHANGELOG.md: edit title to add version info: `## [Backdoor.4.18981][YYYY-MM-DD]`

## Step Github releases

- [ ] Go to Github releases,  create release based on the new created tag and publish
  - Choose a tag: v4.18981
  - Target: master
  - Set as the latest release
  - Title: `proxmark3-v4.18981`
  - Description:
```
Release v4.18981
Nickname "Backdoor"
```

## Step Homebrew updates

- [ ] update homebrew repo, file `proxmark3.rb`
  - with a SHA256 sum of the file `v4.18981.tar.gz`
  - with updated list of standalone modes

## Step package maintains

- [ ] make a list of new standalone modes, so when we alert package maintainers they have a sporting chance of adding them

doegox, 2024-09-09:
some issue with opensuse-leap, will look tomorrow

doegox, 2024-09-10:
Damn, Ubuntu 18 saw a buffer overflow that no other compiler saw. Easy to fix but I'd like to understand why it saw it and how to add such check to newer setups...
```
[-] CC src/cmdhflist.c
In file included from /usr/include/string.h:494:0,
                 from src/cmdhfict.c:21:
In function 'memcpy',
    inlined from 'diversify_mifare_key' at src/cmdhfict.c:151:5,
    inlined from 'derive_mifare_key' at src/cmdhfict.c:189:5,
    inlined from 'CmdHfIctReader' at src/cmdhfict.c:199:12:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:34:10: error: '__builtin___memcpy_chk' writing 8 bytes into a region of size 6 overflows the destination [-Werror=stringop-overflow=]
   return __builtin___memcpy_chk (__dest, __src, __len, __bos0 (__dest));
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In function 'memcpy',
    inlined from 'diversify_mifare_key' at src/cmdhfict.c:151:5,
    inlined from 'derive_mifare_key' at src/cmdhfict.c:189:5,
    inlined from 'CmdHfIctReader' at src/cmdhfict.c:203:12:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:34:10: error: '__builtin___memcpy_chk' writing 8 bytes into a region of size 6 overflows the destination [-Werror=stringop-overflow=]
   return __builtin___memcpy_chk (__dest, __src, __len, __bos0 (__dest));
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

```c
    uint8_t diverse[MIFARE_KEY_SIZE]; // = 6
    diversify_mifare_key(uid, diverse);
```

```c
    uint8_t output[8];
...
    memcpy(app_key, output, sizeof(output));
```
(should be `memcpy(app_key, output, MIFARE_KEY_SIZE);`

iceman1001, 2024-09-10:
I saw that one on 22.04 ,   or was it 24.04,     because I fixed it on my laptop.   Strange that the newer builds never saw it...   or maybe it just warning and we only stop on error?

iceman1001, 2024-09-10:
and since coverity changed their tool last year in december,  I haven't gotten it to work.   Hence its red.

doegox, 2024-09-10:
> I saw that one on 22.04 , or was it 24.04, because I fixed it on my laptop. Strange that the newer builds never saw it... or maybe it just warning and we only stop on error?

We err on warnings, all compilation flags are same.
only diff is gcc v7.5 (!!) vs gcc v14...

doegox, 2024-09-10:
ok I'm done with docker tests, few fixed pushed.
your turn for the missing platforms :)

iceman1001, 2024-09-10:
Release done.

closed 2024-09-10 by iceman1001