RfidResearchGroup/proxmark3 #2495: colors module used by recover_pk is very dead

ZeroChaos-, 2024-09-10:
When running the tests I noticed something odd:

```
recover_pk test                         [ FAIL ] ❌ 
Execution trace:
Traceback (most recent call last):
  File "/var/tmp/portage/net-wireless/proxmark3-9999/work/proxmark3-9999/tools/recover_pk.py", line 11, in <module>
    from colors import color
ModuleNotFoundError: No module named 'colors'
```

Colors in pypi is an alpha 0.1 release from 2005 with a dead homepage.  https://pypi.org/project/Colors/#history
Is this the right library?  I wouldn't have noticed but it wasn't package in gentoo and I was going to package it, but it's really unmaintained at this point.  Would you accept a PR to use a different library such as ansicolors?

iceman1001, 2024-09-10:
```# Requirements:
# python3 -m pip install ansicolors sslcrypto
```

closed 2024-09-10 by iceman1001

ZeroChaos-, 2024-09-10:
I wasn't asking how to install ansicolor, I was asking if I can convert this code to use it instead of the ancient dead library in use now. 
 https://github.com/RfidResearchGroup/proxmark3/blob/master/tools/recover_pk.py#L11

marcelbarbulescu, 2024-09-10:
PM3 is already using the [`ansicolors`](https://github.com/RfidResearchGroup/proxmark3/blob/master/tools/requirements.txt) package, which provides the `colors` module.

ZeroChaos-, 2024-09-10:
My system must be broken, I will fix that. Thanks.

iceman1001, 2024-09-10:
Read my post again and you find your answer.