RfidResearchGroup/proxmark3 #2496: hf iclass sam SAM select failed bug

thesle3p, 2024-09-10:
**Describe the bug**
HF iclass sam gives a SAM select Failed error when attempting to read from a HID sam chip using a SIM to Smartcard adapter

**To Reproduce**
Steps to reproduce the behavior:
1.
 ```
pm3 --> hf iclass sam -v
[=] ISO7816-3 ATR : 3B 95 96 80 B1 FE 55 1F C7 47 72 61 63 65 13 
[+] SAM (Grace) detected
[#] failed to receive from SIM CARD
[!] ⚠️  SAM select failed
```
2.
```
[usb] pm3 --> smart info
[=] --- Smartcard Information ---------
[=] ISO7816-3 ATR... 3B 95 96 80 B1 FE 55 1F C7 47 72 61 63 65 13 
[=] Fingerprint..... IClass SE Processor (Other)
[=] https://www.hidglobal.com/products/embedded-modules/iclass-se/sio-processor

[=] ATR
[=]     - TA1 (Maximum clock frequency, proposed bit duration) [ 0x96 ]
[=]     - TD1 (First offered transmission protocol, presence of TA2..TD2) [ 0x80 ] Protocol T0
[=]     - TD2 (A supported protocol or more global parameters, presence of TA3..TD3) [ 0xb1 ] Protocol T1
[=]     - TA3: 0xfe
[=]     - TB3: 0x55
[=]     - TD3 [ 0x1f ] Protocol T=15
[=]     - TA4: 0xc7
[=]     Historical bytes ( 5 )
[=]     00: 47 72 61 63 65                                  | Grace

[=] D/F (TA1)
[=]     - Di 32
[=]     - Fi 512
[=]     - F  5.0 MHz
[=]     - Cycles/ETU 16
[=]     - 250000.0 bits/sec at 4 MHz
[=]     - 312500.0 bits/sec at Fmax (5.0MHz)
```
3.
```

[usb] pm3 --> smart reader
[=] ISO7816-3 ATR... 3B 95 96 80 B1 FE 55 1F C7 47 72 61 63 65 13 
[=] Fingerprint..... IClass SE Processor (Other)
[=] https://www.hidglobal.com/products/embedded-modules/iclass-se/sio-processor
```
**Expected behavior**
hf iclass SAM should be able to read the tags provided

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
```
 -  [ Proxmark3 RFID instrument ]

 [ Client ]
  Iceman/master/v4.18994-suspect 2024-09-10 15:40:28 4ecb7df89
  compiled with............. GCC 11.4.0
  platform.................. Linux / x86_64
  Readline support.......... present
  QT GUI support............ present
  native BT support......... present
  Python script support..... present
  Lua SWIG support.......... present
  Python SWIG support....... present

 [ Proxmark3 ]
  device.................... RDV4
  firmware.................. RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... absent

 [ ARM ]
  bootrom: Iceman/master/v4.18994-suspect 2024-09-10 15:40:28 4ecb7df89
       os: Iceman/master/v4.18994-suspect 2024-09-10 15:40:28 4ecb7df89
  compiled with GCC 10.3.1 20210621 (release)

 [ FPGA ] 
 fpga_pm3_hf.ncd image 2s30vq100 2024-02-03 15:12:20
 fpga_pm3_lf.ncd image 2s30vq100 2024-02-03 15:12:10
 fpga_pm3_felica.ncd image 2s30vq100 2024-02-03 15:12:41
 fpga_pm3_hf_15.ncd image 2s30vq100 2024-02-03 15:12:31

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 72% used )

```
 - hw status
```
[#] Memory
[#]   BigBuf_size............. 39588
[#]   Available memory........ 39588
[#] Tracing
[#]   tracing ................ 0
[#]   traceLen ............... 10
[#] Current FPGA image
[#]   mode.................... fpga_pm3_hf_15.ncd image 2s30vq100 2024-02-03 15:12:31
[#] Flash memory
[#]   Baudrate................ 24 MHz
[#]   Init.................... ok
[#]   Memory size............. 2 mbits / 256 kb
[#]   Unique ID (be).......... 0x2A9FB7DF230C69D5
[#] Smart card module (ISO 7816)
[#]   version................. v4.13 ( Outdated )
[#] LF Sampling config
[#]   [q] divisor............. 95 ( 125.00 kHz )
[#]   [b] bits per sample..... 8
[#]   [d] decimation.......... 1
[#]   [a] averaging........... yes
[#]   [t] trigger threshold... 0
[#]   [s] samples to skip..... 0 
[#] 
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | n/a | n/a | 
[#]     long leading reference |  29 |  17 |  18 |  50 |  15 | n/a | n/a | 
[#]               leading zero |  29 |  17 |  18 |  40 |  15 | n/a | n/a | 
[#]    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override.... std    ( follow standard )
[#]   [b] BCC override........ std    ( follow standard )
[#]   [2] CL2 override........ std    ( follow standard )
[#]   [3] CL3 override........ std    ( follow standard )
[#]   [r] RATS override....... std    ( follow standard )
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed................... 500ms
[#]   Bytes transferred.............. 305152
[#]   Transfer Speed PM3 -> Client... 610304 bytes/s
[#] Various
[#]   Max stack usage......... 3520 / 8480 bytes
[#]   Debug log level......... 1 ( error )
[#]   ToSendMax............... 6
[#]   ToSend BUFFERSIZE....... 2308
[#]   Slow clock.............. 29910 Hz
[#] Installed StandAlone Mode
[#]   LF HID26 standalone - aka SamyRun (Samy Kamkar)
[#] Flash memory dictionary loaded
[#]   Mifare.................. 1888 / 2047 keys
[#]   T55x7................... 124 / 1023 keys
[#]   iClass.................. 28 / 511 keys
```

 - data tune

```
help             This help
-----------      ------------------------- General-------------------------
clear            Clears various buffers used by the graph window
hide             Hide the graph window
load             Load contents of file into graph window
num              Converts dec/hex/bin
plot             Show the graph window
print            Print the data in the DemodBuffer
save             Save signal trace data
setdebugmode     Set Debugging Level on client side
xor              Xor a input string
-----------      ------------------------- Modulation-------------------------
biphaserawdecode Biphase decode bin stream in DemodBuffer
detectclock      Detect ASK, FSK, NRZ, PSK clock rate of wave in GraphBuffer
fsktonrz         Convert fsk2 to nrz wave for alternate fsk demodulating (for weak fsk)
manrawdecode     Manchester decode binary stream in DemodBuffer
modulation       Identify LF signal for clock and modulation
rawdemod         Demodulate the data in the GraphBuffer and output binary
-----------      ------------------------- Graph-------------------------
askedgedetect    Adjust Graph for manual ASK demod
autocorr         Autocorrelation over window
convertbitstream Convert GraphBuffer's 0/1 values to 127 / -127
cthreshold       Average out all values between
dirthreshold     Max rising higher up-thres/ Min falling lower down-thres
decimate         Decimate samples
envelope         Generate square envelope of samples
grid             overlay grid on graph window
getbitstream     Convert GraphBuffer's >=1 values to 1 and <1 to 0
hpf              Remove DC offset from trace
iir              Apply IIR buttersworth filter on plot data
ltrim            Trim samples from left of trace
mtrim            Trim out samples from the specified start to the specified stop
norm             Normalize max/min to +/-128
rtrim            Trim samples from right of trace
setgraphmarkers  Set the markers in the graph window
shiftgraphzero   Shift 0 for Graphed wave + or - shift value
timescale        Set cursor display timescale
undecimate       Un-decimate samples
zerocrossings    Count time between zero-crossings
-----------      ------------------------- Operations-------------------------
asn1             ASN1 decoder
atr              ATR lookup
bitsamples       Get raw samples as bitstring
bmap             Convert hex value according a binary template
crypto           Encrypt and decrypt data
diff             Diff of input files
hexsamples       Dump big buffer as hex bytes
samples          Get raw samples for graph window ( GraphBuffer )
```
**Additional context**
Add any other context about the problem here.

bettse, 2024-09-12:
I tried this and had a similar problem until I upgraded my sim module firmware, so that is something to try if you haven't already.

thesle3p, 2024-09-12:
Yeah I updated the sim module firmware and the issue still persists.

iceman1001, 2025-02-20:
Is this still an issue?

Closing meanwhile