RfidResearchGroup/proxmark3 #2500: Unable to read tag with hf_legic.lua

fprackwieser, 2024-09-11:
When I try to read a tag with the hf_legic script I always get this error.

```
> rt
[+] TYPE: MIM256 card (234 bytes)
[+] Reading tag memory.
[+] Saved 256 bytes to binary file `C:\ProxSpace\pm3/hf-legic-temp-********.bin`
[+] Saved to json file `C:\ProxSpace\pm3/hf-legic-temp-********.json`

256 bytes from C:\ProxSpace\pm3/hf-legic-temp-AdKqeYSu.bin loaded
create virtual tag from 256 bytes

Tag-Type: SAM
1 Segment(s) found
256 bytes for Tag processed
[-] Error - can't find `hf-legic-temp-********.eml`
[-] error - C:\ProxSpace\pm3\proxmark3\client\luascripts/hf_legic.lua:769: bad argument #1 to 'os.remove' (string expected, got nil)

[+] finished hf_legic

[usb] pm3 -->
```

iceman1001, 2024-09-12:
Aha,   we don't generate `.eml` files any more.   

The script needs to be updated

jmichelp, 2024-11-06:
@fprackwieser can you confirm the script works for you now that PR #2615 has been merged?

You may surface other bugs but reading should work :)

iceman1001, 2025-06-08:
Ping,   but closing meanwhile