RfidResearchGroup/proxmark3 #2504: Changes for Gentoo to compile correctly.

drforbin, 2024-09-13:
I thought I'd post this for Gentoo users.

The following I had to do to get the latest release to compile.

```
diff --git a/armsrc/desfire_crypto.c b/armsrc/desfire_crypto.c
index 7a30d4ff3..c93b4590a 100644
--- a/armsrc/desfire_crypto.c
+++ b/armsrc/desfire_crypto.c
@@ -31,7 +31,7 @@
 #include "crc32.h"
 #include "crc.h"
 #include "crc16.h"        // crc16 ccitt
-#include "printf.h"
+//#include "printf.h"
 #include "iso14443a.h"
 #include "dbprint.h"
 #include "BigBuf.h"
diff --git a/common/iso15693tools.c b/common/iso15693tools.c
index 5f6dd7567..2e0711013 100644
--- a/common/iso15693tools.c
+++ b/common/iso15693tools.c
+++ b/common/iso15693tools.c
+++ b/common/iso15693tools.c
+++ b/common/iso15693tools.c
+++ b/common/iso15693tools.c
+++ b/common/iso15693tools.c
@@ -35,13 +35,19 @@ char *iso15693_sprintUID(char *dest, uint8_t *uid) {
     if (uid) {
 #ifdef HAVE_SNPRINTF
         snprintf(dest, ISO15693_SPRINTUID_BUFLEN,
+            "%02X %02X %02X %02X %02X %02X %02X %02X",
+                 uid[7], uid[6], uid[5], uid[4],
+                 uid[3], uid[2], uid[1], uid[0]
+                );
+
 #else
         sprintf(dest,
-#endif
-                 "%02X %02X %02X %02X %02X %02X %02X %02X",
+            "%02X %02X %02X %02X %02X %02X %02X %02X",
                  uid[7], uid[6], uid[5], uid[4],
                  uid[3], uid[2], uid[1], uid[0]
                 );
+
+#endif
     }
     return dest;
 }
```

Please let me know if you'd like me to post a pull request on this.

DF

iceman1001, 2024-09-13:
This functions is one of the more odd ones yet so simple. 
We use `snprintf`  across the client and arm side its available in `nprintf.c`  

yet this function uses a DEFINE to mix in both. 
Wouldn't it be better to just use `snprintf`  and include nprintf.h in  b/armsrc/desfire_crypto.c ?

drforbin, 2024-09-13:
The point is the way the function is defined using the (#define) spread across (if's) is considered non portable and the compiler was  bitching.  As for the other it seems there were multiple includes of printf and defines of it's prototype so again a yuk! So what do you think should be done?

iceman1001, 2024-09-13:
As a first step I think we should use `snprintf`  across the source code.   
The different printf implementations on armsrc is a next step.

drforbin, 2024-09-13:
Also I know this is off topic...BUT where can I find all the switches for proxmark3 commands?

drforbin, 2024-09-13:
Ok. What shall I do?

iceman1001, 2024-09-14:
if you up to it,  swap out the sprintf calls across the code?  
If not, just make a PR with your suggested fix.  its simple enough but doesn't really solve the original problem

drforbin, 2024-09-14:
Ok iceman. FYI great code here. Let me see how much time I have and i'll get around to it. 
Gotta think over weekend. Oh, this is alittle off topic. But what does the square brackets mean around SAK? and the '()'s around the other data in list(s)?

iceman1001, 2024-09-14:
Not sure about great code,   but we are trying to make some order into it.  Many chefs has added their flavor to it and its "messy" still.    But we try. 

I guess you mean the output from 14a info etc.
```
[+]  UID: 04 59 50 92 0C 6A 80   ( double )
[+] ATQA: 00 44
[+]  SAK: 00 [2]
```

The [2]  is actually the response code from our "anticollision"  process,   It can be either 0, 1, 2 or 3
The code comments says
```
    /*
        0: couldn't read
        1: OK, with ATS
        2: OK, no ATS
        3: proprietary Anticollision
    */
    ```

In short  1, 2 and 3 are all positive responses,   
```
1 = 14a Card with ATS support
2 = 14a Card with no ATS support
3 = 14a Card that doesn't support standard iso14443-3 anti-collision
```

we normally don't add response code in the output we prefer textual feedback.   But since this function is a complex nightmare with more and more additions over the years,  it was added for maintainers to get a what our function returns when encounter strange tags in the wild.

So you can happily ignore that piece of information

drforbin, 2024-09-14:
Thxs so much!!

what about in hf 14a list where it says i.e. (40 (7)) for example ?
Is that seven bits per byte? xfered?

iceman1001, 2024-09-14:
in the trace list  that particular situation indicate number of bits.   It isn't super accurate,  its more of a guess work by looking at time spent receiving data.

closed 2024-09-14 by iceman1001

drforbin, 2024-09-14:
Total number of bits for that data item?
Tell me what you'd like me to do with the errors above?
replace that function?

iceman1001, 2024-09-14:
estimate of total bits for short frames in ISO14443A would be best description. 

> if you up to it, swap out the sprintf calls across the code?
> If not, just make a PR with your suggested fix. its simple enough but doesn't really solve the original problem

drforbin, 2024-09-14:
Ok....thxs. Where can I read about short frames in the standard?

iceman1001, 2024-09-15:
Read a ISO14443-A document ?