RfidResearchGroup/proxmark3 #2547: Build error  on hitags

JohnDoeIsKIng, 2024-10-01:
Dear community,

last week I encountered an error during building. The error evolved on  the target hitags.
Here 's the build log:
`compiler version:  arm-none-eabi-gcc (15:6.3.1+svn253039-1build1) 6.3.1 20170620
[-] GEN fpga_version_info.c
[-] CC fpga_version_info.c
[-] GEN obj/fpga_all.bit.z
[-] GEN obj/fpga_all.o
[-] CC start.c
[-] CC iso15693.c
[-] CC ../common/iso15693tools.c
[-] CC thinfilm.c
[-] CC lfops.c
[-] CC lfsampling.c
[-] CC pcf7931.c
[-] CC ../common/lfdemod.c
[-] CC lfadc.c
[-] CC ../common/lz4/lz4.c
[-] CC legicrf.c
[-] CC legicrfsim.c
[-] CC ../common/legic_prng.c
[-] CC ../common/hitag2/hitag2_crypto.c
[-] CC hitag2.c
[-] CC hitagS.c
hitagS.c: In function 'hts_select_tag':
hitagS.c:1158:13: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
             key_le = *(uint64_t *)packet->key;
             ^~~~~~
hitagS.c:1160:13: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
             uint64_t state = ht2_hitag2_init(reflect48(key_le), reflect32(tag.data.s.uid_le), reflect32(*(uint32_t *)rnd));
             ^~~~~~~~
hitagS.c:1252:13: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
             uint64_t state = ht2_hitag2_init(reflect48(key_le), reflect32(tag.data.s.uid_le), reflect32(*(uint32_t *)rnd));
             ^~~~~~~~
cc1: all warnings being treated as errors
../common_arm/Makefile.common:118: recipe for target 'obj/hitagS.o' failed
make[1]: *** [obj/hitagS.o] Error 1
Makefile:167: recipe for target 'armsrc/all' failed
make: *** [armsrc/all] Error 2
`

`john@pc:~/Githubs/proxmark3$ cc --version
cc (Ubuntu 10.3.0-1ubuntu1~18.04~1) 10.3.0
`

Any hints to solve that are appreciated. 
Best regards

JD.

iceman1001, 2024-10-01:
No idea.   GH Actions doesn't show any build errors or warnings. 

Have you cleaned your env?   `make clean; make -j`    ?

JohnDoeIsKIng, 2024-10-02:
Greets iceman,

yes I have cleaned and build new.
The issue seems to relate to a compiler standard, see here:

[https://stackoverflow.com/questions/7630150/dereferencing-type-punned-pointer-will-break-strict-aliasing-rules](url)

I will try to build a fix and come back.
Regards

JD.

JohnDoeIsKIng, 2025-01-29:
Dear community,

I've gabled around a little bit more with the issue. First I've updated to gcc 13 which didn't solve the issue.
Then, I analyzed the error itself and found that the issue comes from the respective warning being treated as an error (-Werror=strict-aliasing). 

The I've tried to find the respective code in the Makefiles of the armsrc target and tried to change that with no luck.
Also I've tried to deactivate the hitagS-target in the armsrc.c but that led to further issues.

So I'd like to ask if it is possible in any way to deactivate that particular target to avoid the breaking of building.
The issue was that one:

hitagS.c:1228:13: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]

Regards,

JD.

iceman1001, 2025-01-29:
`hitagS.c:1228`   line 1228 in the hitagS.c file 

Which is 
`uint64_t state = ht2_hitag2_init(REV64(key_le), REV32(tag.data.s.uid_le), REV32(*(uint32_t *)rnd)); `

looks like that type casting of rnd is the culprit.

JohnDoeIsKIng, 2025-01-30:
Greets Iceman,

thanx for replying. 
Can you provide a hint how to improve that code line ?
Or should I continue trying to delete that warning treated as an error of my local clone?
Regards,

JD.

commit ffbf0339 (2025-02-20): fix #2547 - compilation error on arm-none-eabi-gcc 15.6 for  error: dereferencing type-punned pointer will break strict-aliasing rules,   by making a u32 we dont get that any longer
commit 44221012 (2025-02-21): fix #2547 - compilation warning error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
closed 2025-02-20 by iceman1001